ArduPilot/ardupilot #3785: Alternative QUAD-plane setup

raptortrias, 2016-03-22:
Considering an alternative setup where the two front motors are used in both QUAD configuration and FIXED wing configuration. The transition would involve tilting these two motors with servos for transition to fixed wing

Is it possible to map output for twin in fixed wing and the same 2 motors used for quad for quad. 

Would use servo controlled tilt to transition from the 2 front motors from quad configuration (facing up) to forward movement in fixed wing?

Thanks

marcmerlin, 2016-03-29:
I like your idea, but you're now back in the experimental aircraft situation.
The transition from plane to quad would be more difficult because you would be without thrust of any kind when the servos are moving, and it would not work with the current code. In other words, you would have to write that code, or convince someone to, most likely with a paid contract :)
As a hint, though, I know of no one having done this successfully, Tridge, Google Wing, and others, picked a quad plane design, because it's just more reliable and easy to implement.

My recommendation is to close this issue in a couple of weeks if no one picks it up and you're not going to work on it yourself, or have someone do it for you.

magicrub, 2016-03-29:
For the record, independently controlled twin engine fixed wing is not yet supported, that would probably be a pre-req to this.

geofrancis, 2016-04-05:
i was thinking about this and about how to get apm to fly a tilting quadplane 

connect tilting motors to autoflaps function so they come down as the speed decreases.

the only part that would have to be changed in software would be at what point do you switch your 4 motors from quadcopter motor out to the airplane motor out. a mix would be even better but harder to implement.

you could have it so that its only quadcopter mode when the motors are at flap position 1 and 2 but aircraft motor out when the flaps are up.

DavidIngraham, 2018-01-09:
I believe this has been supported since 3.8.0. In fact, FoxTech has a commercial product that demonstrates this capability: https://www.foxtechfpv.com/nimbus-1800-long-range-vtol.html

auturgy, 2018-01-09:
Thanks David.
This configuration is now supported.

closed 2018-01-09 by auturgy