ArduPilot/ardupilot #3788: MAVLink: Create an enum for flight modes

WickedShell, 2016-03-22:
Currently we report the current flight mode as a custom_mode in the heartbeat, which is fine, but the different vehicles (copter/plane/rover) all use different values for what each mode is. We should have an enum in the ardupilotmega.xml that indicates what flight mode we are in, this way the GCS isn't dependent upon hard coding that  on plane custom mode of 4 is acro while on copter that is guided. If we could move this all to a common enum that all vehicles share this would help on the GCS side a lot.

Because of the intrusiveness of the change this might be best rolled out with MAVLink 2, or some other major GCS breaking update.

rmackay9, 2016-03-23:
a fine idea!

WickedShell, 2018-03-05:
Enum was created and accepted, the decision was that vehicle code does not reference it, but otherwise is complete.

closed 2018-03-05 by WickedShell