ArduPilot/ardupilot #3800: Can't connect to Telem2 on Arducopte 3.3

OpenGelo, 2016-03-26:
Hi, I want to connect my companion computer to pixhawk via telem2. Before this, I had tried to connect to Telem1 and it works. I also tried on latest arduplane. There is no issue with telem2. This is the second pixhawk board that i tried to connect to a companion computer. Both had same issue. So, I think maybe it's a bug for arducopter 3.3, isn't it?

Thanks

mboland, 2016-03-26:
Have you tried setting your serial parameters for Telem2?

OpenGelo, 2016-03-28:
yes I have @mboland , but only set the baudrate

mboland, 2016-03-28:
You will also need to set what data is sent and how.
All I did was copy the parameter values from the main telem1 to the telem2 and all worked as expected.
In my case I was setting up Graupner telemetry and wanted it to use the telem2 rather than 'Y' connected off telem1.

OpenGelo, 2016-03-28:
@mboland All parameters of telem1 is in Serial1, right?

the protocol is GCS Mavlink
I just set two values in serial2 (baudrate and protocol) like this picture http://ardupilot.org/copter/_images/MP-Serial2_protocol.png

rmackay9, 2016-03-28:
You may need to also set the BRD_SER2_RTSCTS parameter to zero.

I'm going to close this as really we don't want support issues in our issues list because it distracts from the confirmed bugs and enhancements requests.

closed 2016-03-28 by rmackay9