ArduPilot/ardupilot #3807: I2C connection Question

sgh055, 2016-03-28:
Hello 

I have a question. 

Pixhawk support some sonar sensor. But I want to use MLX90614(temperature sensors) using I2C.

Is this possible connection?

If you know way to connect please tell me some link or data. thank you.

Jasonsiu, 2016-03-28:
what do you want to do by MLX90614?Althold?obstacle avoidance?

jpkh, 2016-03-28:
Long ago ardupilot was using thermopiles but that support has been dropped as currently we are using gyros/accels to determine orientation. @sgh055 I would suggest you to write your own library support for this thermopile sensor.

sgh055, 2016-03-28:
@Jasonsiu I would just measure the air temperature for embers detection

@jpkh Then I'd get where did the information on i2c library of ardupilot?

jpkh, 2016-04-18:
@sgh055 naturally from here on [Ardupilot](https://github.com/ArduPilot/ardupilot) repository

closed 2016-04-18 by jpkh