ArduPilot/ardupilot #3812: RFE: AP 3.5.2: RTL and other auto modes should have an option to turn off motor when above target altitude

marcmerlin, 2016-03-28:
My motor glider doesn't need motor power to fly at all times, and there are 2 clear cases
1) try to maintain altitude (cruise) but in an updraft
2) RTL or AUTO and above target altitude

Taking the easy case of RTL above target altitude, it consistently uses motor for me.
Like here, target RTL altitude is 100m, I'm at 145m, and motor is at 43%
[image]

Turning off power to 0% may not be desirable for all airframes, but definitely for motor gliders if you're well above target altitude.
There is of course the trick of how much airspeed you want to keep before your turn the motor back on, so maybe some
FBW_GLIDE_AIRSPEED
If airspeed is above that, and we're above target altitude, keep airspeed at this setting and glide down.
If airspeed is decaying below that speed, pitch down further
If target altitude (below) is about to be reached and airspeed is around FBW_GLIDE_AIRSPEED, turn the motor back on.
Sounds fair?

marcmerlin, 2016-03-28:
@WickedShell asked for an example with log, so the screenshot above was taken from this flight log: http://marc.merlins.org/tmp/7_RTL_USES_MOTOR_ABOVE_ALT.BIN
Hope this helps.

marcmerlin, 2016-03-29:
By the way, the plus from adding a new FBW_GLIDE_AIRSPEED is that it would default to 0, and disable the feature of turning the motor off in AP controlled flight modes. That way, nothing changes by default, and people with good glide ratios, can choose to enable it.
Does it sound like a good idea?

geofrancis, 2016-04-02:
my plane would turn off the motor when it was decending in RTL.

marcmerlin, 2016-04-05:
@geofrancis : do you have throttle nudge enabled?
I do, and RTL turns the motor on even if my throttle is 0%

magicrub, 2016-04-05:
Marc, you must have some param set somewhere strange. If you have plenty of
airspeed, and descending in a loiter, then the motor should be off.

marcmerlin, 2016-04-05:
Ah, I am not hitting the descending into a loiter case. I'm hitting the "flying to the RTL point, but my glide ratio is 10:1, and I don't need motor to arrive there with sufficient altitude"
Has anyone gotten their plane to glide in the direction to the RTL point (not in loiter above the RTL point)?

geofrancis, 2016-04-05:
try calibrating your ESC.

magicrub, 2016-04-05:
What does that do?

marcmerlin, 2016-04-08:
So I did a test yesterday evening, flew over the RTL point, switched to RTL, and the plane did indeed circle down without using the motor, so that's good.
Now, the logic just needs to be a bit smarter and try to glide down to the RTL location without motor power if some heuristics say it can, and FBW_GLIDE_AIRSPEED is set.

geofrancis, 2016-04-09:
i think i had throttle nudge enabled so if i hit RTL and i wanted to fly at minimum airspeed i would have to put the throttle to 0. 

i would have thought with it disabled and using the airspeed sensor it would default to your cruse airspeed.

marcmerlin, 2016-04-11:
@geofrancis thanks for those pointers. Now, should TRIM_THROTTLE only be relevant if I'm trying to maintain altitude?
If I'm in RTL and I need to travel to a point, while losing altitude, should it still apply? (I think not)
I have: TRIM_THROTTLE   40.000000

I could try setting that to 0, but is that the right/safe thing to do, just so that RTL can go down with throttle off? If so, I'll give it a shot.

geofrancis, 2016-04-19:
i would lower it a little first rather than jump to 0

Buljan, 2017-02-01:
Hi guys,

I''m doing a high altitude balloon airplane launch, and the project is on a standstill until we figure out how to RTL the plane with no motor running- while being 20km above the Home position.  
This topic is the best discussion that i found, is there any progress? 

Or, do you guys have a alternate solution?

Thanks,
Josip

geofrancis, 2017-02-03:
just set trim throttle to 0 and it will try and cruse with no throttle if you dont have an airspeed sensor.

Buljan, 2017-02-04:
Hey, thanks for the reply, but i do have an airspeed sensor installed. :/

any tips?

geofrancis, 2017-02-04:
if you have an airspeed sensor then it should keep the throttle at zero as long as you are above minimum air speed.

magicrub, 2017-02-08:
We've had other people do balloon drops and I don't recall this being a
problem. Does anyone else know they that did it? Can we contact any of them?

geofrancis, 2017-02-08:
if your FBWB_CLIMB_RATE is too low then it will run the throttle to stop the plane decending too fast.

IamPete1, 2021-08-06:
This can be achieved with soaring support. (also lots of other cool stuff)

https://ardupilot.org/plane/docs/soaring-4_1.html

closed 2021-08-06 by IamPete1