ArduPilot/ardupilot #3813: 'waypoint reached' message needs translation

magicrub, 2016-03-28:
The current human readable string announcement for reaching a mission item (nav or cmd) is not very human readable. Lets upgrade that to be more useful.

Current:
Executing nav command ID # 16"

Proposed:
"Reached mission item 3 of 10 now executing LOITER_TIME"

Discuss.

rmackay9, 2016-03-29:
An argument against this is that the flight controller shouldn't be so involved in the UI.  The GCS is much better placed to do this.

hsteinhaus, 2016-03-29:
What about generally adding message ids for EVERY text message sent by APM, and maybe later replacing the text messages completely? This approach would save a lot of telemetry bandwidth and ROM space and also would make internationalization/localization easier. 

I did some front end development recently  ( https://github.com/hsteinhaus/ArduPilot_Taranis ), on the remote end of a very narrow-bandwidth telemetry link. There is no chance to send all these bulky messages in plain text, but transferring just a message id (plus maybe some arguments to be format-printed into the message) would be fine. 

Currently some telemetry converters (like the one described here in the Wiki here: http://ardupilot.org/copter/docs/common-frsky-telemetry.html ) basically parse the message received by MAVLink, look up an ID for it, transmit  the ID  over the telemetry link and convert it back. What a wasting of resources.

WickedShell, 2016-03-29:
There was an issue floating around for that @hsteinhaus but I can't seem to find it now. However there are two problems with that as we still need to add some additional content, and you can't expect the GCS to know the text in advance. The previous conclusion was that we still need to store the text strings on the AP and allow the GCS to query them if it doesn't know how to display it. For the most part we should be trying to move as much into pure mavlink as possible.

That said this is something that we should work on in a separate issue.

magicrub, 2016-03-29:
This was discussed on the weekly dev call and we decided to add the "3 of 10" type thing but leave off the LOITER_TIME type label. We can revisit that later. That is best handled by the GS

peterbarker, 2017-12-19:
Relates to #7291

peterbarker, 2020-04-11:
So of course we've done the opposite of what was described here.  We now have:

```
    gcs().send_text(MAV_SEVERITY_INFO, "Mission: %u %s", cmd.index, cmd.type());
```

We could move to
```
    gcs().send_text(MAV_SEVERITY_INFO, "Mission: %u/%u %s", cmd.index, num_commands, cmd.type());
```

peterbarker, 2021-07-30:
@magicrub do you think the output is sufficiently good now?

magicrub, 2021-07-30:
I'll have to check master

IamPete1, 2021-08-30:
@magicrub bump on that check ;)

magicrub, 2021-08-30:
BUMP ACK.

magicrub, 2024-05-07:
Agreed, they're much better now. Closing.

Start of all mission items:
"Mission: 3 WP" or "Mission: 4 LoitTime" or "Mission: 5 Land"

```
AP_Mission::start_command()
GCS_SEND_TEXT(MAV_SEVERITY_INFO, "Mission: %u %s", cmd.index, cmd.type());
```

closed 2024-05-07 by magicrub