ArduPilot/ardupilot #3819: Flight mode channel in Plane and Copter differs.

pepevalbe, 2016-03-29:
For people that uses both ArduCopter and ArduPlane it might be confusing that the flight mode channel is not the same.
* In copter it is always Channel 5. 
* In plane it is configurable with FLTMODE_CH parameter. Its default value is 8, however this is not specified in the parameter description. 

Shouldn't we change this default value to 5 so they match? Or at least update FLTMODE_CH description to specify the default channel.
 
If people agree I can make a PR for this and change de wiki accordingly.

marcmerlin, 2016-03-29:
Channel 5 for plane doesn't work for most people who have at least 2 independent aileron servos (2nd aileron is almost always channel 5).
Then if you add flaps, that's channels 6 and 7, so that's why flight mode on plane is 8 by default.
It's easier to change copter to 8, than plane to 5 if you need to make them match.

But I'm not sure if it's worth the hassle to existing users to change the FM channel, I think most would be upset. I just use a different channel for plane and copter.

I agree however that having the flight mode channel configurable on copter, would be a good thing.
I'll change your issue to reflect that.

pepevalbe, 2016-03-29:
The 2nd aileron is disabled by default so people who want 2 independent aileron movements need to configure an aditional parameter anyway. They could use other channel diferent than 5 or change the flight mode channel parameter as well. It don't think it is a bad idea to modify the default value. 

My original thought was that both flight mode channels should match to avoid confusion for people who are new to ArduCopter coming from Arduplane (and viceversa). Once you know they are different being able to make them match does not really help.

marcmerlin, 2016-03-29:
@pepevalbe you may be right, but almost all arduplane users (AFAIK) have mode switch on channel 8. Changing them all to channel 5 isn't going to go well, as in it's a non starter IMO.
Making copter configurable to another channel has no impact on users, but I'm assuming you understand that if we were also going to just change all copter users to channel 8 for mode switch, they wouldn't be happy either.
As a result, I think we're stuck with the status quo, but at least making the mode switch channel configurable would be good.

R-Lefebvre, 2016-03-29:
Keep in mind, the motor on/off function on helicopters is currently hard-coded to CH8.

I think it makes sense to create the parameter for copters, make it the same name as Plane, but make it default to Ch5 so that people who do nothing, get the same result they have now.

This is generally in the direction of unifying the code, so it's a good thing.

R-Lefebvre, 2016-03-29:
Yes, I've tried twice to bring in a system giving us complete flexibility on ALL channel assignments, and it was declined twice.  So I'll just let somebody else take a crack at it.  I think a big part of the reason we are stuck, is the concept of pure channel pass-through on failsafe for airplanes.

marcmerlin, 2016-03-29:
you're right that passthrough for planes is important, but in this case not relevant.
I have my MS set to CH12 since channels 1-8 are the important ones.
For copters, it's passthrough is irrelevant, so we can put any channel anywhere.

Then there is the issue that you may not want MS on a channel higher than what your TX can send (maybe only 6 channels for copters?).
So honestly, just having MS be configurable on copter, would be a good start IMO. Having one default MS channel that will work for all, is a bit more doomed IMO, but I also don't think it's that important as long as it's configurable.
I'm personally more annoyed at stuff like variables called
ARSPD_ENABLE
vs
ARSPED_FBW_MAX
I mean SPED is just like SPEED and SPD got into a car accident and got turned into SPED, or whoever was using scrabble letters to name variables ran out of 'E's :)

pepevalbe, 2016-03-30:
I don't see any problem on new plane installs having different MS channel than old planes. In the future most of them would have the new value.
As far as I know the real reason why MS default is CH8 is because of the hardware failsafe on APM1. New boards doesn't have hardware failsafe, right? so we don't need to keep it.

R-Lefebvre, 2016-03-30:
There's a lot of decisions being made, that start and stop at "We can't break people's setups and make them reconfigure."  And it's really holding back development of some very needed functionality.  This ties in directly with full flexibility on output functions.  We keep getting stuck on this, and it's just going to be more complicated when we have CANbus ESC's, serial servos, etc.

marcmerlin, 2016-03-31:
@R-Lefebvre note that this issue is now explicitly about making the channel changeable on copter, as opposed to hardcoded :)
Wouldn't that be progress already? 
From there, people can argue about defaults, but let's not get the main point derailed: hardcoded channels in ardupilot, should go, be it in copter, or plane.

R-Lefebvre, 2016-03-31:
Yes, I understand.  But I wanted to point out that, this problem with channels is much bigger than just mode channel selection.  I'd prefer to see a clean sheet rewrite of the entire channel system, instead of adding another patch fix on top of previous patch fixes.

pepevalbe, 2016-03-31:
I agree with Robert. My only intentention was to modify the default value in plane.

WickedShell, 2016-03-31:
I think the answer is pretty clear at the moment that the default value
will not be changing in plane as it will break a lot of people's setups,
and not enough is gained by doing so.

Frief84, 2016-05-04:
Yes but it would pritty cool if you could change the flight mode channel.. because some 
RC-Remotes unable to change the channel assignment for certain switches ( Spektrum DX7 old) 
especially if you use the satellite

rmackay9, 2018-01-27:
This is in master now and will go out with Copter-3.6.
https://github.com/ArduPilot/ardupilot/commit/89674482a717bd280c594c408c6feec7d569fac9

closed 2018-01-27 by rmackay9