ArduPilot/ardupilot #3820: GPS Configuration issue 3.5.2

Jman841, 2016-03-29:
I have had the same issue on both planes i have updated from 3.5.0 to 3.5.2.

When trying to arm I recieve the error:

"Prearm: GPS1 has not been fully configured"

I have never had issues with the GPS in the past and have not changed any parameters away from stock for the GPS. Also, these are original 3DR Ublox units and after looking over the params I don't see anything weird.

marcmerlin, 2016-03-29:
I think I've seen the same issue too, and it went away by just waiting.
I believe it meant that it was still waiting for a better GPS lock.
Would that match what you experienced?

Jman841, 2016-03-29:
We waited for a quite a while and had 12+ sats and still it did not clear.

magicrub, 2016-03-29:
That is a new feature where we are verifying that the GPS was configured correctly. There is an arming check that you can bypass for it but if it's failing it means there may be something wrong. The GPS takes about 10-15 seconds to fully configure so sometimes it takes a simple retry but if it persists then check your cables, maybe data is only going in one direction.

Jman841, 2016-03-29:
Tried multiple times but does not clear and waited a long time. Also, we get full DGPS lock without issue. All configurations are stock and using the 3DR Ublox GPS. 

I believe there is an error with the check itself as this is on two different pixhawks that I have tried. Or possibly an error with the stock settings?

magicrub, 2016-03-29:
Reopened. Do you have some logs @WickedShell can take a look at?

http://ardupilot.org/plane/docs/parameters.html#arming-check-arm-checks-to-peform-bitmask

use ARMING_CHECK=4094 to check for everything except the GPS_config

Jman841, 2016-03-29:
sure, give me a sec. I'll post them shortly

Jman841, 2016-03-29:
Here's a link to the .tlog.

I tried downloading the dataflash log but for some reason my plane has only 1 data flash log from weeks ago. Is there anything in the latest update that would stop the logging on the pixhawk as well?

https://www.dropbox.com/s/gtq5bc75koq102i/2016-03-29%2013-14-41.tlog?dl=0

WickedShell, 2016-03-29:
I need to see the parameters, and when I extracted the parameters from the tlog you were missing a large portion of the GPS parameters. The only ones there were:
GPS_AUTO_SWITCH 1.000000
GPS_MIN_DGPS    100.000000
GPS_MIN_ELEV    -100.000000
GPS_NAVFILTER   8.000000
GPS_SBAS_MODE   1.000000
GPS_TYPE        1.000000
GPS_TYPE2       0.000000

One of the other changes to the GPS driver is that the aircraft will send a mavlink message that tells the GCS what type of GPS is connected, and we should see it say u-blox followed by the hardware and software version. You either connected to late to see it, or you weren't connected long enough (this tlog is only 5 seconds long). The dataflash log would be more helpful for that since it would be guranteed to record, and should have the complete list of parameters for me to review.

The problem should be that you have enabled a feature that you're GPS is not capable of (IE asking for RAW data when your GPS is not capable of it, or enabling a second GPS that isn't connected (we can see from the current parameters this isn't the case). Or that you have a bad wire from the pixhawk to the GPS and the pixhawk can't transmit messages to the GPS, (or at least not reliably enough to pass).

I strongly suspect this is just a support issue, but I'm going to hold off on closing it at the moment as the tlog is missing GPS parameters that should be there (probably just GCS/mavlink dropouts/to short a log).

Jman841, 2016-03-29:
I will try again, but for some reason there is no dataflash logs. Do you want my .param file?

WickedShell, 2016-03-29:
Slightly embarrassed as I should have noticed this earlier but "Prearm: GPS1 has not been fully configured" means that the second GPS instance hasn't been configured (it's 0 indexed which I realize now is very poor consistency as we have the parameters for the second GPS appended with 2. I will get a fix for that in for the next release.

Anyways it's the second GPS that is a problem, and in the param file that Jman sent me he had GPS_TYPE2 = 1 which means it was looking for a GPS, and until it finds one it isn't considered configured. Since no one said they were doing multi-gps stuff I'm operating on the assumption that's the cause, and am closing it at this point in time.

closed 2016-03-29 by WickedShell

Jman841, 2016-03-29:
That makes sense. Thank you.

WickedShell, 2016-03-29:
Actually the if configured but not detected then fall is as designed. The
concern is you might think you have multiple GPS setup and this is our only
chance to catch that.