ArduPilot/ardupilot #3823: Waf name convention

khancyr, 2016-03-29:
Hi,

Not sure if it is an issue or not but actually when building with waf for rover, we got ardurover.px4 in ardupilot/build/px4-v2/bin, when it should be APMrover2-v2.px4 .

Actually, it is the same thing for copter....

Is that right ? or the naming convention will change with waf builds ?
Adding a page about waf on the new wiki could be usefull !

gmorph, 2016-03-29:
Hi Gustavo.  Can you check this please?
Thanks, Grant.

guludo, 2016-03-31:
@khancyr Yes, the binaries names and paths are using a different convention.

History regarding paths: https://github.com/ArduPilot/ardupilot/pull/3481 (maybe it's a good idea to follow the older versions of the PR :-))

About the binaries names: only those of the "main" products where changed. The PR that introduced that is https://github.com/ArduPilot/ardupilot/pull/3518 . There haven't been much discussion on that matter, but I personally think the new names are nicer and more consistent with what is used for common binaries.

Regarding the creation of a wiki page, I agree it would be useful. Let's create an issue for that. In the meantime, people can use https://github.com/ArduPilot/ardupilot/blob/master/README-WAF.md .

closed 2016-11-09 by khancyr

magicrub, 2016-11-09:
Please provide a reason for closing.

khancyr, 2016-11-10:
see Guludo answer