ArduPilot/ardupilot #3825: There should be an ARM override

marcmerlin, 2016-03-29:
I have this problem all the time: debugging on my bench, and I can't arm.
First, I can't arm because of GPS lock, and then because my airspeed is not healthy because my airspeed sensor is not plugged in, because it's on my wing, which is not attached to my plane :)

More generally, you may be on the field and get an error that you understand and want to overrule.

In other case, I'm forced to get a ground station software, set ARMING_CHECK to 0, and then back to 2030, which I cannot even set with tower on android anyway, so then I have to go get my laptop.
Giant pain in the rear on the field.
On my desk, my big fear is to forget ARMING_CHECK left at 0.

Proposed solutions:
1) ground station software should have an "override arm"
2) If you have BRD_SAFETYENABLE, the safety switch is kind of useless. It would be great to have it so that if you try to arm, and BRD_SAFETYENABLE is 0 (i.e. disable the switch for arming), then the switch can be used as for override arming

I'm open to other solutions, but I do think we need some override arm. Yes, it's a shotgun, but we don't all have to aim it at our foot :)

squilter, 2016-03-29:
> which I cannot even set with tower on android anyway

Tower lets you change params.  What's the problem here?

> 1) ground station software should have an "override arm"

We want to make sure it difficult to override the arming checks.

WickedShell, 2016-03-30:
I'm inclined to say the ability to use the param is the way you can force
the override.

rmackay9, 2016-03-30:
Not sure if it's helpful but copter has different arming criteria depending upon the flight mode.  So for example, it's nearly always possible to arm in "Stabilize" which is a (nearly) completely manual flight mode.  Doing it this way of course opens up the possibility that the home location isn't set so RTL may not work correctly.  Anyway, just adding info on the copter method in case that helps in some way.

marcmerlin, 2016-03-30:
@squilter tower does not let me change param back to 2030 (it's a bug with the UI), and more generally, it takes quite a bit of effort to do it.
Right now, the end result is that I disable them, and may not re-enable them, which is a lot worse than a one time conscious disable, which wouldn't sticky.
Am I the only one who does motor and prop testing on the bench, and has to go through this arming override disabling/enabling dance every time?
I have already forgotten to re-enable the arming checks before going to fly, and that's very bad :(

magicrub, 2016-05-27:
@marcmerlin Let me summarize: You want a bench-able motor test without changing params, or at least hard to do on a tablet that may be forgotten to be done.

Do you have any suggestions how on the work-flow? Maybe a mavlink command which will spin the motors for a moment (0.1 to 5 seconds)?

marcmerlin, 2016-05-27:
Correct. I have 5 motors and 15 sets of props, and honestly the only way to really know what you're going to get is to test them on the bench.
The first thing that comes to mind, and that would also help in other cases, is to allow holding the arming switch for let's say 5 seconds and it would arm the motors even if the arming checks don't pass.
Or if you prefer, pushing the switch quickly 5 times, or whatever you think is best.
As another option if you don't like this, simply allow an override arm option through mavlink, which would again arm even if pre arm checks fail.
How do either sound?

marcmerlin, 2016-05-27:
But to answer your direct question, a mavlink command that spins the motor is not what I'm looking for. 
1) sometimes I just need to arm without passing the checks
2) a bench motor test involves me controlling power level from a stick, and looking at multiple graphs. Turning the motor 5 seconds or whatever is not the right thing to do. Sometimes I may even abort the motor test if 10% power doesn't sound good.

magicrub, 2016-05-27:
I like those ideas, we just need to be super careful that they don't get used on accident or abused in a way that reduces safety.

marcmerlin, 2016-05-27:
Agreed on "used by accident", but "abused in a way that reduces safety", you just can't.
We're talking about making the bypassing of safety checks, safer. The mere fact that you're bypassing safety checks means that if used improperly it can reduce safety. But right now, you already do that if you need it, however you have to turn all safety checks off and remember to turn them back on later (which sadly I have already forgotten to do)

magicrub, 2016-05-27:
Well I certainly agree that the current scheme of turning off the checks is
very bad and leads to problems like you've had. Something should be
available to test the motor just before every flight without major param
changes and/or a dozen GCS clicks. Sounds like a topic for the next dev
call.

marcmerlin, 2016-05-28:
Thanks, to be clear this would fix my issue of testing the motor, but more generally I think it makes sense to use the same mechanism to allow force- arming the plane if all the tests don't pass and the pilot needs to arm once and potentially fly anyway. If you don't agree with that last bit, allowing motor arm on the bench only still would be a good thing for testing.

WickedShell, 2018-07-17:
The MAVLink support for this went in with #4891. Bug your preferred GCS authors for buttons on new FW versions.

closed 2018-07-17 by WickedShell

marcmerlin, 2018-07-17:
awesome, thank you.