ArduPilot/ardupilot #3837: Plane: Parachute ALT_MIN should be not above home, but above ground

naktinis, 2016-03-31:
At the moment CHUTE_ALT_MIN parameter prohibits parachute deployment when `relative_altitude() < parachute.alt_min()`. However, `relative_altitude()` is altitude above home. This is incorrect, for example, when you start a terrain-following mission on a hill. As you go downhill, your `relative_altitude()` will decrease and you might not be allowed to deploy the parachute.

I couldn't find an existing value to use instead of `relative_altitude()`, so here are my suggestions:

1. Create a new method `Plane::relative_ground_altitude()` (could be partly inspired by `Plane::calc_altitude_error_cm`).
2. Rewrite `Plane::relative_altitude()` to be relative not from home, but from the ground (since I assume this method is heavily used, this might be tricky).

naktinis, 2016-05-02:
I am happy to produce a pull request for this. However, as it affects altitude calculation on the Plane object, it would be nice to get an approval from someone in the main development team.

magicrub, 2016-05-02:
Option 1 sounds good, but I think we have member functions that utilize
terrain, don't we? May be nice to utilize lidar there too.

naktinis, 2016-05-03:
Here are the getter methods in `altitude.cpp` that utilize terrain altitude:
 - `Plane::relative_target_altitude_cm` – return relative to home target altitude in centimeters. Used for
  altitude control libraries
 - `Plane::calc_altitude_error_cm` – return error between target altitude and current altitude
 - `Plane::height_above_target` – return the height in meters above the next_WP_loc altitude
 - `Plane::lookahead_adjustment` – work out target altitude adjustment from terrain lookahead

From what I see, none of them return "_current altitude above terrain (or home, if terrain not enabled)_".

magicrub, 2016-05-03:
Seems the terrain check is missing from relative_altitude(). You could also pass home to above_location_current()

naktinis, 2016-05-04:
I've looked at all calls of `relative_altitude()` (apart from parachute code):

    ArduPlane.cpp:    gcs_send_text_fmt(MAV_SEVERITY_INFO, "Landing approach start at %.1fm", (double)relative_altitude());
    Attitude.cpp:     fabsf(relative_altitude()) < g.ground_steer_alt);
    GCS_Mavlink.cpp:  relative_altitude() * 1.0e3f,    // millimeters above ground
    altitude.cpp:     float correction = relative_altitude() - rangefinder_state.height_estimate;
    landing.cpp:      hgt_afe = relative_altitude();
    sensors.cpp:      height = relative_altitude();

The one that needs most attention is the call in `Plane::tecs_hgt_afe(void)` (in landing.cpp), others are either non-critical or already have an explicit terrain check (which means they could be easily refactored).

As I don't want to rush with this, I propose the following plan:

1. Write a new function `Plane::relative_ground_altitude()` (or do you like `Plane::relative_terrain_altitude()` better?). If terrain following is enabled it would call `terrain.height_above_terrain(...)`, otherwise it would use `relative_altitude()`.
2. Use `relative_ground_altitude` in places that currently use `relative_altitude` + explicitly incorporate terrain data.
3. (Optional) Rename `relative_altitude` to `relative_home_altitude`.
4. At a later time (or when this gets more investigation or approval), `relative_home_altitude` can get dropped altogether – the idea being that `relative_ground_altitude` should always be at least as precise as `relative_home_altitude` and practically fit for the purpose.

What do you think?

magicrub, 2016-05-04:
I think there's an important difference between above home and above
ground. That case is when you're landing and a different location than
where you took off from. Personally, alt above home is somewhat useless if
terrain and rangefinder data is available. HOWEVER, rangefinder data is
noisy (it sees trees and buildings) so we don't want to use it for
navigation unless it's specifically useful such as in the case of following
terrain features and landing.

For now, lets add a new relative_ground_altitude which uses rangefinder,
else terrain, else baro-home. At a later time we can rethink
removing/renaming the currently popular relative_altitude().

naktinis, 2016-05-05:
Originally for this task I did not intend to add rangefinder data (only terrain). I wanted to have a method that would give the parachute code the best estimate of the distance to the ground. For this puspose, I think, rangefinder data is not necessary (also because of the noisiness you've mentioned).

Also, if the method is named `relative_ground_altitude`, incorporating rangefinder could be misleading because, as you've said, it would show not only ground points, but also trees, buildings, pylons etc.

magicrub, 2016-05-05:
closed via https://github.com/ArduPilot/ardupilot/pull/4045

closed 2016-05-05 by magicrub