ArduPilot/ardupilot #3840: waf: create a wiki page

guludo, 2016-03-31:
Hi all,

This is a followup of https://github.com/ArduPilot/ardupilot/issues/3823 .

Best,
Gustavo Sousa

Naterater, 2019-01-07:
Done: http://ardupilot.org/dev/docs/building-setup-windows10.html#build-with-waf

auturgy, 2019-01-08:
> We do have instructions here for building the code on MacOSC and Linux.
>
> It would have been nice to have a better description of what the issue is above... I think it's a complaint that we've changed the naming convention of the final binary when moving to waf so that the name on it's own no longer clarifies what platform it's for. The directory where the binary appears though does make it clear what platform it's for.
>
> Basically I'm not sure if we really have a problem with the naming of the binaries... I suspect we don't.

we don’t have a waf user guide external to BUILD.md which is a bit out of date: add to that the fact that the command to build a specific target is dynamic (ie with the hwdef system, the build script looks for the name of the board you want to configure and build, rather than having a list of build targets in the script), and it leads to confusion for newcomers imho.

Regards,

James

rmackay9, 2019-01-08:
Is BUILD.mp out of date?  Tridge updated it in mid Nov.. i'm not aware of any issues with it.  I'm not sure how the build instructions ended up in BUILD.md rather than on the wiki but in any case, one place is better than two so let's not duplicate it... although you may not have been suggesting that in any case..

IamPete1, 2021-08-08:
BUILD.md looks quite good to me, so I'm going to close, of course we can always improve documentation, but that should be tracked in a issue over on the wiki repo.

closed 2021-08-08 by IamPete1