ArduPilot/ardupilot #3848: Parachute Pre-Check

Elios007, 2016-04-03:
HI, we perform a pre-check for everything but we are still not able to do it for the parachute yet as the ardupilot will understand it as the mission has finished. Before each flight, we check that the parachute door will open, so it should be included inside the pre-check procedure.

magicrub, 2016-04-03:
I can't think of a good way to pre check a chute, do you have any suggestions?

khancyr, 2016-04-04:
Hi,

There are two problems here : 
How the autopilot will catch that the parachute is open when using normal servo ? We got no feedback...
Do we need the check when using pyro parachute? 

Here is how I procede for parachute check. When booting, pixhawk put pwm to 0. So it open my servo parachute. Then it goes to close pwm (1500) and close the parachute. I know that it is work and can fly safe!

closed 2016-11-30 by Elios007