ArduPilot/ardupilot #3849: Panic Button

Elios007, 2016-04-03:
If we can create a Panic Button to let the plane gain certain predefined altitude (example +100m) and also to bypass some parameters as Tecs-climb (if it was low below 3m/s). 

Sometimes you may face some situation that you need to gain altitude very fast while you are at any mode, instead of trying to switch to stabilize mode to fly the plane, you just switch this panic button on the RC radio and the plane will go up by +100m and will start to do loiter as example.

iskess, 2016-04-04:
If you already have the controls in your hand, when not just flip to FBWA and pull full back on the stick with full throttle? Don't use Stabilize mode or you could over control the aircraft.

marcmerlin, 2016-04-05:
You can use a geofence that's very big (so it never triggers) and put a minimum geofence altitude, as well as a (higher) geofence return altitude.
You then hit your panic button, which turns geofencing on, and the airplane will climb to the geofence return altitude. In theory, it's supposed to climb there pretty quickly, but if you're preventing it with climb speed parameters, or maximum motor allowed, maybe you want to revisit that.

marcmerlin, 2016-04-05:
As an easier way, I also have an RTL switch as a panic button, it'll recover, and then climb to a safe altitude.

peterbarker, 2021-07-30:
There were several valid options presented here - and we now have  scripting available which allows for this sort of personalised experience.  Closing this.

closed 2021-07-30 by peterbarker