ArduPilot/ardupilot #3850: RCMAP_ROLL = 8 -> roll stabilization don't work

davidkraus13, 2016-04-04:
Hi,
I needed to control roll from channel 8, so I use RCMAP_ROLL parametr. after that change I re-calibrate radio and in MP everything looks OK. But when I try to lift of after one or two seconds copter turn over. I try how motors behave without props, and it behaves very strange (minimal or almost no reaction for RC input). After change RCMAP_ROLL = 1, everything works well.

Using pixhawk with exterrnal mag+gps, running arducopter 3.3.2 - hexa X

rmackay9, 2016-04-04:
Can you put a log file somewhere?  normally we don't answer support questions here in github but we can make an exception for this one because you're doing something that I'm not sure I've ever seen anybody try.  I've seen people use rcmap to move around channels 1~4 but I've never seen it used for input from a much higher channel.  I don't immediately know of a reason why it wouldn't work though.

davidkraus13, 2016-04-04:
[2016-01-07 18-12-28.zip](https://github.com/ArduPilot/ardupilot/files/202351/2016-01-07.18-12-28.zip)
Here is older log file from crash, I hope I found right file.
I put it here because it looks for me like bug in code, we try on 2 pixhawk units, and setting make 2 people and it behave same.

closed 2021-02-08 by IamPete1