ArduPilot/ardupilot #3851: Plane:Parachute can't manual release

jinchengde, 2016-04-04:
We set the plane automatic armed without safety switch (gas engine plane), but the parachute can't release and MP prompt "Parachute: no flying".
ArduPlane set the soft_armed each update_AHRS there:
`hal.util->set_soft_armed(arming.is_armed() &&
                   hal.util->safety_switch_state() != AP_HAL::Util::SAFETY_DISARMED);`
so, could we remove the check of safety_switch?

jinchengde, 2016-04-29:
That is not about Safety_Switch, but mention to `bool Plane::is_flying(void)
{
    if (hal.util->get_soft_armed()) {
        // when armed, assume we're flying unless we probably aren't
        return (isFlyingProbability >= 0.1f);
    }

    // when disarmed, assume we're not flying unless we probably are
    return (isFlyingProbability >= 0.9f);
}
`
I will check the code for that

magicrub, 2016-04-29:
In a stalled aircraft situation, is flying is unreliable. The parachute code should not be using is flying. I'll take a look at this soon regarding removing it.

magicrub, 2016-06-07:
fixed. Removed is_flying check (https://github.com/ArduPilot/ardupilot/commit/ef3d17cc4c4c5269c15f873641cb1fefd25149d8) and added manual ground tests (https://github.com/ArduPilot/ardupilot/commit/263931b4b1c78760580488a622beb6e2b8987c22)

closed 2016-06-07 by magicrub

jinchengde, 2016-06-08:
@magicrub Thanks Tom!

jinchengde, 2016-06-08:
@magicrub 
Hi Tom， could you explain why we need to check 

> auto_state.started_flying_in_auto_ms > 0 && auto_state.takeoff_complete

Does that mean user could only manual open the parachute in auto mode? if we takeoff in fbwa mode or others, then could we release the parachute?

magicrub, 2016-06-08:
Yes, that it what it means and it should change to be for all modes. I'll
fix it today.  Thanks for checking

magicrub, 2016-06-08:
> auto_state.started_flying_in_auto_ms > 0 && auto_state.takeoff_complete

checks to see if you're *ever* flown before (in auto) but doesn't care if you're flying right now or not. Changed to test for any mode (https://github.com/ArduPilot/ardupilot/commit/65e0e31a7a62341305cb97a4dab7e6ae396f3aeb). Sometimes I forget that people fly around manually. I prefer 100% auto :) Much more reliable and repeatable.

jinchengde, 2016-06-13:
Nice job!