ArduPilot/ardupilot #3852: Plane: Parachute always released before motors are off

naktinis, 2016-04-04:
How we want this to work (I assume):
 - parachute release sequence initiated (`parachute.release()`)
 - throttle suppressed (`suppress_throttle` code checks if parachute released initiated)
 - parachute triggered (servo or relay triggered, `parachute.released()` starts returning true)

How it currently works:
 - parachute release sequence initiated (`parachute.release()`)
 - throttle suppressed only when `parachute.released()` is true (that is, it waits until servo or relay is actually triggered, see `suppress_throttle` definition)
 - parachute triggered (servo or relay triggered, `parachute.released()` starts returning true)

Proposed fix: add new method `parachute.release_inititated()`, `suppress_throttle` should check if `parachute.released_initiated()` instead of `parachute.released()`.

Correct me if my assumption is wrong. Otherwise, I'm happy to contribute a fix.

magicrub, 2016-04-04:
There is probably about 100ms between those checks, are you saying that the throttle is giving you problems and you need that extra 100ms ?

naktinis, 2016-04-04:
I'm saying that, from what I understand, throttle is currently not suppressed until the parachute is actually deployed. And I think we want to first suppress the throttle and only then deploy the parachute.

naktinis, 2016-04-22:
Should be fixed by df922dacfa.

closed 2016-04-22 by naktinis