ArduPilot/ardupilot #3853: Plane: Potential problem with altitude offset

naktinis, 2016-04-04:
I was trying to compare how `relative_target_altitude_cm` and `adjusted_relative_altitude_cm` are incorporating `ALT_OFFSET` and it seems odd.

Target altitude is calculated like this (before rangefinder correction):

    target_relative_to_home_altitude = target_altitude.amsl_cm - home.alt + alt_offset_cm

However, in `Plane::adjusted_relative_altitude_cm(void)`:

    current_relative_to_home_altitude = current_loc.alt - home.alt - alt_offset_cm

Notice the sign difference in front of `alt_offset_cm` in both equations. Why so?

To expand a little more, I make the following assumptions:
 - `target_altitude.amsl_cm` - target altitude above mean sea level, calculated as target changes and is never changed or adjusted afterwards (that is until the target itself changes, of course), may be off by some offset
 - `home.alt` - home altitude above mean sea level, calculated once (before the flight) and is never changed or adjusted afterwards
 - `current_loc.alt` - autopilot altitude above mean sea level, dynamic value of where autopilot "thinks" it is, may be off by some offset
 - If we know that autopilot "thinks" it is lower than it actually is, we add a positive `ALT_OFFSET`

magicrub, 2016-04-19:
Thanks @naktinis . I think this has gone unseen because ALT_OFFSET is so rarely used. In fact, I'm in favor or removing it in favor of the more global GND_ALT_OFFSET

WickedShell, 2017-02-10:
As far as I can tell ALT_OFFSET only exists to shift mission items independently of the baro at this point, so this can be safely closed.

closed 2017-02-10 by WickedShell

magicrub, 2017-02-10:
And to make this me clear, I'm in favor of moving that params to MIS_Alt_offset.