ArduPilot/ardupilot #3865: Pixracer r14 / Current master: Hott (Graupner) PPM does not work

GervaisK, 2016-04-06:
My r14 board has probs with Hott (Graupner) PPM .

Calibration works fine, but RC Receiver message appears quite often and Failsafe ´s getting engaged. Probs with 3.3.4 (Pixracer stable), as well as with current master. 4/6/2016

And no, this is not a RC hardware problem. I´ve got several nicely working GR-16

R-Lefebvre, 2016-04-06:
How do you know it's not an RC hardware problem?  You might have other GR-16 systems working well, but you might have one bad unit.  I've got 20 FrSky Rx, and I had one of them that would consistently shut down after a few minutes in the air triggering an RC Failsafe.

One idea to check, I've found that some of my systems go into RC Failsafe if the Tx is *too close* to the Rx while I'm doing bench testing.  Needs to be at least 1-2 feet away.

rent4yj, 2016-11-10:
i have same problems. 
gr-16l not a work in pixracer but in pixhawk is good work 
i don't know what a happen ! 
so i test futaba r7008sb. it's good work 
i use 3.5-dev firmware build in ubuntu !! 
@GervaisK are you solve this problem??

tonylm, 2017-02-12:
No I have same problem, It's not a Radio problem it's firmware problem because don't work with ardupilot, but work with Px4 firmware

IamPete1, 2021-08-30:
Any update? We support Hott telm now so I suspect this had a fair bit of testing recently.

https://ardupilot.org/copter/docs/common-hott-telemetry.html

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1