ArduPilot/ardupilot #3868: Motor Test is broken in trunk (3.4)

rrr6399, 2016-04-07:
We updated a couple of different Hexacopters this week with the 3.4 firmware from 4/2/2016. When we tried to run the motor test, it refused to run due to being at too low an altitude.Arming and everything else works fine.

rmackay9, 2016-04-08:
Ok, thanks.  I suspect we have to set the desired-spool state to be "throttle-unlimited".

magicrub, 2016-05-27:
There was a recent bugfix by Tridge for quadplane (https://github.com/ArduPilot/ardupilot/commit/9933069679caf5b6e77018093d780707e45130d4). Maybe it's related?

OXINARF, 2016-05-27:
@rrr6399 Can you confirm this is still happening?

rmackay9, 2016-05-30:
I've just tested the motors test on a quad with the latest beta and it seems to work.  The report that "altitude is too low" is very odd.  I can imagine that "throttle too low" could be a report.
In any case, it seems to work as far as I can tell so without any recent feedback that it's broken I'm going to close this issue.

closed 2016-05-30 by rmackay9

rrr6399, 2016-05-30:
I'll test it out this week once I get my hexacopter working again. I imagine if its working for Randy it should work for me as well.

rrr6399, 2016-06-04:
Looks like it does work, although the initial power setting in Mission Planner isn't high enough to move the motors.