ArduPilot/ardupilot #3873: Instances GPS

InigoCastro, 2016-04-07:
Hello people!  I am a new user of the firmwre code and I am having some problems to understand a concept related to the GPS.
So, I am currently trying to grasp the code of the gps module and I have no idea neither What is nor what role plays a INSTANCE of the gps, and It seems important since it appears constantly in the code

I would welcome some orientation about the concept.

Thank you in advance.

rmackay9, 2016-04-08:
Questions should probably go to our gitter channel.  https://gitter.im/ArduPilot/ardupilot

In any case, the instance allows the caller to specify which GPS they want to talk to.  So we support more than one GPS although at least for Copter, we don't recommend that because we don't handle the switch over from one GPS to another very well at the moment.

WickedShell, 2016-04-09:
In general use the primary insurance field, out just call the versions of the functions that don't take an instance argument. 

Closing at this isn't an issue, this kind of thing is best asked on gitter or drones-discuss

closed 2016-04-09 by WickedShell