ArduPilot/ardupilot #3875: AP 3.5.2's RTL does not seem to recover from inverted flight, and crashed

marcmerlin, 2016-04-08:
Log: http://marc.merlins.org/tmp/2_Inverted_RTL_Crash.BIN
Video of the crash: https://youtu.be/wFwTneYeNY4?t=12m17s

Things start at 10:44 after takeoff, 12:16 in the osd video offset
I was flying manually, lost sight, but sadly I was almost fully inverted by the time I engaged RTL.
However, when it takes over, very weird things happen, the video speaks a thousand words:
RTL pitches to 100 degrees (still inverted) and apparently uses the elevator to pitch towards the ground, making the ground scroll backwards.

It's hard to say if RTL would have recovered with more altitude, but it did weird things and not the right steps to recover right away. Therefore, looks to me like a bug since it could have flown out of it, and didn't.

Channels/Mixes:
o CH01 : right Aileron A
o CH02 : Elevator      E
o CH03 : Throttle      T (notrim so that we can use the trim slider for other stuff)
o CH04 : Rudder        R
o CH05 : left aileron
o CH06 : R flap
o CH07 : L flap
o CH08 : OSDCmg: up 12V+cams off, mid: 12V on, down: GUIDED. 2 mode flips: OSD rotate
o CH09 : FPV Camera L/R (S2)
o CH10 : free
o CH11:  RSSIQual from RX
o CH12 : Ardupilot Mode Switch (6 position switch mod), lost model finder plugged in Pixhawk

marcmerlin, 2016-04-09:
@WickedShell @magicrub could you review my analysis and say if it's sound?

Ok, a big thanks to @tridge who helped teach a man to fish :) and helped me out analyse this.

So here's the summary, plane starts in MANU, mostly inverted (my fault, lost visual on the plane)
I switch RTL right away as this happens.
RTL manages to roll the plane back to almost 100 degrees (from 150 or so), and correctly knows to use the elevator backwards to help fix pitch down. Sadly as it gets to 100 degrees, the elevator now mostly acts as a rudder and causes a backwards spin (left turn while banked right). This creates a huge angle of attack on the wings as the plane is very much not pointing in the direction of travel. I'm convinced that this mostly stalled the wings and especially the ailerons.

What goes wrong then is this:
1) AP continues applying motor power when airspeed goes past FBW_MAX. Tridge and I agreed that it should be safe to cut motor power if airspeed is that high, no matter what the condition in a throttle controlled mode. It may not avoid the crash, but will give more time to avoid it and make fewer pieces of plane if it does impact the ground

2) More importantly, AP does not understand knife edge (90 degree roll), does not use the rudder as an elevator and sadly continues to use a lot of elevator while the plane is close to 90 degree bank.
The video and graph make it clear that AP was recovering from the bad attitude until it reached 100 degrees, and then stopped making progress (or made very little progress).
My guess at this point is that being rolled right while making a left turn in a spin with such elevator control, caused a big angle of attack on the wings and negated the ailerons trying to turn the wing around. In other words, the rules for stalling are currently based on airspeed when really they should be based on angle of attack, but AP doesn't have an AOA sensor.
In this case it though it couldn't be stalled with such airspeed, except I'm convinced it mostly stalled the wings, stopping the ailerons from working and flipping the plane back up.

3) Just to make things worse, there is well meaning code that reduces aileron deflection as airspeed increases. This is usually good because the faster you go the less aileron deflection you need. But that's not true if your wing is partially or fully stalled. In that case, you need full deflection to have a fighting chance to override the stall.
Mind you, this is not something you do when you're right side up because you don't fix a stall with ailerons, if you need the ailerons to bring your plane back to a proper attitude, I'd argue that allowing full aileron deflection if the desired roll isn't being attained, could be considered. Notwithstanding the 2 points above, I do believe full ailerons would have allowed the plane to roll back up despite the bad things being done by the elevator.

If I'm correct about the above points, suggested code changes:
1) (suggested by tridge): cut motor power if speed is above FBW_MAX

2) use a cosine function to mix elevator and rudder commands into rudder and elevator as you get closer to 90 degree roll. When I was close to 90 degree roll, the AP should have send 80% of pitch up to the rudder, and only 10% to the elevator (or somesuch). This would likely have avoided this huge backwards turn and had a better chance of keeping nose up with the rudder
This in turn would hopefully not have stalled the wing's ailerons and allowed for the roll to finish

3) The above 2 seem kind of obvious, @tridge did mention this 3rd one can be more tricky due to unintended consequences if we get the conditions wrong.
Is it possible if desired roll is not being achieved to ignore the speed reduces aileron travel rule, and add more ailerons despite speed?
If that's not safe to do always, can we do it if the roll is between 70 and 110 degrees?
While adding too much aileron at high speed can cause air separation and loss of lift, I'd argue that past a certain roll angle, the wings are making very little lift anyway and if we fail to roll away from that state despite an AP command to do so, adding more aggressive ailerons ought to be the thing to do.

marcmerlin, 2016-04-09:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

geofrancis, 2016-04-09:
is there a reason the ruder has not been used up until now? as far as i know its not stabilised in any way?

iskess, 2016-04-09:
The rudder is stabilized, just not by default. You need to activate the yaw controller.

marcmerlin, 2016-04-09:
@geofrancis to be fair, using the rudder would have helped a bit, but it wouldn't have saved this crash. The amount of elevator wrongly applied by AP caused that inverse spin and the ailerons to stall out
@iskess which setting is that? I have these below. Looks like autotune didn't set any and I need to do this by hand, correct?
YAW2SRV_DAMP    0.000000
YAW2SRV_IMAX    1500.000000
YAW2SRV_INT     0.000000
YAW2SRV_RLL     1.000000
YAW2SRV_SLIP    0.000000

Either way, as long as elevator and rudder aren't mixed with a cosine function depending on the plane's roll, this won't make much of a difference for upset recovery like in this case.

iskess, 2016-04-09:
Auto tune doesn't do yaw. 
I agree it probably wouldn't have saved your plane, I'm just responding to the question as to whether there is a yaw control loop. 
For instructions see: http://ardupilot.org/plane/docs/roll-pitch-controller-tuning.html

You need to change _INT and _SLIP

geofrancis, 2016-04-11:
a cosine function mix would also help with acro mode, it would allow the pilot to do level rolls with the auto pilot maintaining pitch using a combination of rudder and elevator.

magicrub, 2016-04-13:
The analysis certainly seems reasonable. Great work @marcmerlin

marcmerlin, 2016-04-14:
@magicrub thanks for looking this over along with the other folks. Note that @tridge did the hard work of the initial review, but he guessed that my ailerons stopped working at high speed (a problem with foam apparently), while looking at the video again and visualizing the very weird reverse spin it was doing, I came to realize that the elevator introduced a huge angle of attack and stalled the wings, and therefore the ailerons.
(current stall protection is of course a bit naive due to lack of AOA sensor, and using airspeed as a guide was totally worthless when your attitude is about 90 degrees roll).

Now of course, the harder work of writing one or multiple fixes for this, remains :)

marcmerlin, 2016-06-07:
Sigh, this bug really sucks, it looks like the same bug crashed my plane a second time, right after I had rebuilt it from this crash :(  
And this time it had almost 100m to recover, but still failed to do so.
https://github.com/ArduPilot/ardupilot/issues/4288 has the details and OSD video
@tridge I know you're busy with quadplane, but is there a chance the AP logic can be improved to stop crashing planes that are in the 80 to 110 degree roll range?

marcmerlin, 2016-06-11:
@tridge Thinking about this bug again, short of doing a better fix like https://github.com/ArduPilot/ardupilot/issues/3925 , could you add a simpler fix to simply not apply elevator for atitude/altitude control if
roll is in 70 - 120 degree range (or even all the way to 135 degrees, i.e. past 45 degrees upside down).
Actually I would even be happy with 80-120 or even 89-120. I just never want the elevator to make the plane spin backwards. Many planes don't fly as well upside down, so I think it's fair to say that AP should very try to correct pitch past 45 degrees upside down before worrying about correcting altitude/attitude.

I think protection on the high roll + upside down case is the most important since both my crashes were due to AP trying to maintain my altitude/atitude in a way that was simply impossible (with the elevator), but on top of not fixing the altitude, it's also stalling the wings, preventing the ailerons from putting the plane right side up and back to a roll of less than 60 degrees, where the elevator would help fix attitude and altitude again.
Considering my last crash happened from 100m, if it's not able to turn the plane back to less than 90 degrees roll from that altitude, I think it's fair to agree that what it's currently doing is just not going to "probably/hopefully work most times" given even more altitude.
https://youtu.be/Q9p985vMqZ4?t=1m15s clearly demonstrates that AP never managed to get back to less than 90 degrees roll before hitting the ground.

marcmerlin, 2016-06-15:
I captured this last comment into a new issue/solution that can be implement as a stop gap for this problem instead of the fancier one that mixes rudder and elevator depending on roll angle

magicrub, 2016-06-21:
possible solution is to reduce elevator pitch proportional to roll. So, when roll=0 then elevator_gain=1. When roll=90 then elevator_gain = 0

tridge, 2016-06-21:
we've discussed it on the call, and decided to do two things:
- when rolled past LIM_ROLL_CD+500 we will linearly reduce elevator response as we approach 90 degrees, with zero elevator at 90 degrees. This should reduce the spiralling backwards that can be induced in your aircraft
- we will look at reversing rudder when inverted. Paul described how high dihedral gliders get an opposite roll effect from rudder when inverted. It is airframe dependent. I'd like to try and reproduce this with XPlane SITL. We'd need a parameter for this.

tridge, 2016-06-23:
@marcmerlin, can you confirm it was your cularis-7 that had the issue recovering from inverted?
I'd like to build a simulation model of an aircraft that behaves like this, and I'd like to know what the airframe is.

tridge, 2016-06-23:
I've created https://github.com/ArduPilot/ardupilot/pull/4365 to address the first point.
I've also now got a rough X-Plane10 model of a Cularis-7 glider that reproduces the issue with rudder producing reverse roll when inverted. That explains a lot of the issue that your plane has in recovering.
We are going to need an option to reverse rudder feed-fowards from aileron when inverted (the bit that comes from KFF_RDDRMIX). 
The main question is what to call this option? INVERT_ROLL_RDDR=0/1 perhaps?

marcmerlin, 2016-06-23:
@tridge I never got my cularis inverted with AP or crashed it as a result, it's only ever been the BFG2600, but the airframes are similar (BFG is just bigger), so I assume they would behave similarly.
So, glad to read that you were able to reproduce this in the sim with a cularis model.
I'm surprised to read that you're seeing issues related to rudder input though, it always seemed to me that feeding elevator to control pitch and altitude when I'm around 90 degree roll is very counter productive.
Also, just flying this in my head, if I'm at 110 degree roll towards the right, wouldn't the rudder try to yaw the nose left which in this case would actually point the nose up, which also just happens to be something that would help maintain pitch up a bit?

marcmerlin, 2017-06-06:
@OXINARF I think this was fixed in 3.7, was it not? (or at least mitigated)
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/release-notes.txt
"Marc Merlin discovered that on some types of gliders that ArduPilot
would not reliably recover from inverted flight. The problem turned
out to be the use of the elevator at high bank angles preventing the
ailerons from fully recovering attitude. The fix in this release
prevent excessive elevator use when the aircraft is beyond
LIM_ROLL_CD. This should help a lot for people using ArduPilot as a
recovery system for manual FPV flight."

closed 2017-06-06 by marcmerlin