ArduPilot/ardupilot #3881: Guided Takeoff Crashes Hexacopter sometimes in 3.4

rrr6399, 2016-04-09:
For firmware versions from at least since February 1/2016 and as recent as 4/2/2016, I've observed 3 crashes on two different S900 Hexacopters when performing the following MAVLINK actions:

1.) Arm in stabilize
2.) Switch to guided mode
3) Send msg_command_long with the following attributes set:
		msg_command_long cmd = new msg_command_long(this.systemId, componentId);
		cmd.target_component = MAV_COMPONENT.MAV_COMP_ID_SYSTEM_CONTROL;
		cmd.target_system = targetSystem;
		cmd.command = MAV_CMD.MAV_CMD_NAV_TAKEOFF;
		cmd.param3 = 0.0f;
		cmd.param4 = 0.0f; // yaw angle not used  currently
		cmd.param5 = (float)latitudeDeg;  // not used currently
		cmd.param5 = 0f;
		cmd.param6 = (float)longitudeDeg; // not used currently
		cmd.param6 = 0f;
		cmd.param7 = (float)2.0;
		cmd.confirmation=0;
		this.sendCommand(cmd);

When the command was issued the copter immediately yawed, rolled and pitched wildly until I disarmed it as shown here:
[image]

There are many other times, that this process has worked perfectly.  The only thing I noticed that may have contributed to the behavior was that the barometer was not zeroed during arming and registered 2.5 meters or so . I'm not sure what the takeoff mode would do if the copter needed to drop in altitude to achieve the takeoff altitude when it was already on the ground. Here's the altitude history:

[image]

Here's the RCOUT history:
[image]

rrr6399, 2016-04-09:
In both crashes, it looks like the copter started spinning in yaw soon after the takeoff command was issued.

rrr6399, 2016-04-10:
BTW, two of the three crashes had the R/C turned off. Don't think that's the problem the takeoff command has worked on my F550 without the R/C turned on. Also, the following fail safes were disabled:  FS_THR_ENABLE, FENCE_ENABLE. One copter had WP_YAW_BEHAVIOR=0 and the other had it set to 2. There appears to be some logic related to yaw where set_auto_yaw_mode(AUTO_YAW_HOLD). 

There was no RC INPUT for yaw registered in the logs as well.

rrr6399, 2016-04-11:
I noticed that even though the target altitude was 2m, the design altitude grew to 4.54 m
[image]

rrr6399, 2016-04-11:
I noticed that barometric and relalt pressure is off by a meter initially. The target altitude starts off at 1 m and then ramps to the actual target of 2m.
[image]

rrr6399, 2016-04-11:
I noticed that the FCS seems to want to yaw after guided mode even though the pilot didn't command any yaw and there should be no yaw angle targeted:
[image]

rrr6399, 2016-04-11:
Here is how the yaw, pitch, roll and altitude history looks for the same copter during a successful guided takeoff:
[image]
[image]
[image]

havarjoh, 2016-04-11:
funny thing - same happens to me and this other guy, only it happens every time - and in gazebo: http://discuss.ardupilot.org/t/gazebo-sitl-drone-cant-controlled/8051
No resolution though.

rrr6399, 2016-04-12:
havarjoh, did you notice the copter yawing hard on takeoff similar to the attached plots?

havarjoh, 2016-04-21:
jup, but looking at the plots I have more extreme values. I almost got it working (no yawing) sending a "level" mavros command before "arm throttle" and "takeoff 2". Only it didn't stop on 2 meters, just kept on rising until it went crazy again (heavy yawing). Think I'll give px4 stack a try, documentation seems better on Gazebo stuff (though not so on other items).

rrr6399, 2016-05-06:
@rmackay9 One pattern I'm seeing is that when channel 4 (yaw) of RCIN is 0 due to the R/C transmitter being turned off, sometimes the yaw rate is maxed out when switching to the guided mode takeoff. Could there be a bug where a RCIN value of 0 for Yaw is being mapped to a maximum Yaw (PWM 0) rather than PWM 500? I have seen these types of issues in the past where a 0 YAW or Pitch value is understood as maximum left yaw or maximum pitch.

It seems to only happen in certain conditions though since I have now seen the problem with an Iris and an S900, but not my F550. I never turn on the R/C with the F550 and have never seen the maximum yaw rate commanded on guided takeoff.

BTW, WP_YAW_BEHAVIOR is typically set to 0 and most of the fail safes are turned offf.

rmackay9, 2016-05-06:
@rrr6399, yes it's quite possible that that's happening.  the desired-yaw (in the ATT) message should make it clear if that's happening.
There's currently two other issues that we're working on that are problems with Copter-3.4 so it's certainly not completely stable.  https://github.com/ArduPilot/ardupilot/pull/4046
I guess I'm just saying that although I haven't investigated this issue you're reporting much yet, it's still on my to-do list and I think we will get to the bottom of it before too long (some time during the beta testing period which is coming soon).

rrr6399, 2016-05-07:
Here's a good example with the desired yaw, actual yaw and RC4 input for the problematic case where the R/C was turned off:
[image]

rrr6399, 2016-05-30:
@rmackay9 Any news on this? I am going to try duplicating it on my F550. The problem is easy to duplicate on an Iris and S900 for far by commanding a guided takeoff with the R/C turned off.

rrr6399, 2016-06-03:
Another variable in the mix of these copters is the type of R/C receiver, which may explain the difference for my F550. My F550 uses the Spektrum AR8000 receiver coupled to the satellite receiver. When the R/C transmitter is turned off, the AR8000 sends the FS PWM of 913 for the throttle and also the last know PWM settings for all the other channels. For instance, YAW would be 1500 rather than 0. 

The S900 uses the Spektrum satellite receiver only, which I believe does not send any FS values. It sends no-data when the R/C transmitter is turned off. I believe the Pixhawk will still detect this as a loss of signal, but all the other channels may be 0 rather than the last known good position. 

I'm not sure how the Iris receiver radio responds to a turned off transmitter, but my guess is that it is similar to the Spektrum satellite receiver.

Also, during arming, my GCS sends the release R/C control message to the Pixhawk to guarantee that the GCS doesn't  still have control over the throttle, yaw, pitch and roll from a previous run (if the operator decided to use the GCS to control those functions.)

govsux, 2016-06-07:
have you tested the RTL being triggered by shutting off the TX with a receiver set to no pulses???

rrr6399, 2016-07-08:
More details are documented in this discussion.

OXINARF, 2016-11-03:
@rrr6399 Like we came to the conclusion in the thread this was caused by you using no RC receiver and not using override values for all channels. You are free to open an issue requesting an enhancement to improve flying without an RC receiver.

closed 2016-11-03 by OXINARF