ArduPilot/ardupilot #3882: Feature Requeset: Autotune for Traditional Heli's

Jman841, 2016-04-10:
While tuning a Heli is not too difficult if you have a lot of experience, for a new user it can definitely be tricky and can be dangerous as with the P value too high on traditional heli's things can go bad very quickly. 

It would be awesome to have an autotune feature for the pitch and roll of Traditional heli's. The Tail is less necessary IMHO as it's fairly easy to tune and if you use values too high it will most likely not result in a crash.

If we can simplify the tuning process it could possibly open up all the benefits of traditional heli's to many users who are turned off by this.

R-Lefebvre, 2016-04-10:
Yeah, would be very hard to do which is why I've never bothered trying.  I'm also pretty skeptical of the result.  Particularly with helis which can seem to be fine hovering at a given P-value, but soon as you descent straight down into your own wash, can light up the oscillation.  So the test would have to be ascending and descending.

I have thought it would be much easier, and useful to have do an auto-tune for the Feedforward which I find more people struggle with.

I should be fairly straight forward actually.  Simple output a certain move to the swashplate, and then observe the fully developed rate.  From that, you can find the deflection necessary for a given rate.

Jman841, 2016-04-11:
the Feed Forward auto tune is definitely a step in the right direction, but, it seems the P value is the most critical and dangerous for many users to tune, however, to be honest I have no idea of the technicalities required for an autotune system, so, i'm just the good idea fairy on this one. But, any tuning that can be done automatically is a step in the right direction IMHO.

Is it possible for the controller to detect when a PID Value is too high? 

Would it be possible to have a dynamic PID for the issues of descending within the rotor wash? The controller could automatically lower the P or increase the I if it detects oscillations?

R-Lefebvre, 2016-04-11:
Yep, all good ideas.  Just need the time and money to do it. :)

bnsgeyer, 2021-02-10:
This issue is being addressed by PR #14619

closed 2022-06-06 by bnsgeyer