ArduPilot/ardupilot #3883: Copter: High vibration event.

geofrancis, 2016-04-10:
in auto modes if you have some kind of hardware failure there is currently no way for the autopilot to detect them and RTL. 

if the aircraft if flying normally in auto and there is a sudden prolonged high vibrations it could trigger RTL or at least flash a high vibration warning message. i know the new mission planner has the little bar graph now for vibrations but that is no use to most users that use mobile devices.

rmackay9, 2016-04-10:
The high vibration warning message would be best if it's a feature of the GCS I think because the UI is not something we can change in ardupilot (which is what this issues list is for).

Tower can display the vibration levels but it's not enabled by default.  The Tower issue list is here.
https://github.com/DroidPlanner/Tower/issues

We could make a high vibration failsafe (i.e. make it trigger RTL) although when the vibes get bad enough to trigger a bad position estimate an EKF failsafe should be triggered but that normally leads to a LAND (because it's not capable of RTL without a position estimate).

Jman841, 2016-04-10:
I think the biggest issue is the effect of high frequency vibrations on the accelerometers. Would it be possible for Ardupilot to recognize when the clipping of this gets too high and switch to only using the gyroscopes for attitude estimation?

This could atleast keep the multicopter from flipping and crashing and then switch to land mode immediately, or, maybe RTL using only the barometer and GPS to get back? While it will most certainly not be as precise without the accels, it's better than the alterantive of the quad flipping and crashing. Or maybe if this failsafe is enabled it will go to a safe altitude (100 meters or more) where the baro drift is not going to be a major issue, and then instead of land at the RTL point it goes down to a hover at 20 meters and waits for the user to take over to land it as i assume it would be difficult to detect when it touches the ground without the accel?

I'm not sure what level of clipping is the breaking point for the accels, but, it is something we could test and figure out for this failsafe.

rmackay9, 2016-04-10:
We've tried a few things to try and make the EKF more resistant to vibration levels with limited success.  One of those things was similar to your suggestion - make the EKF ignore accel values if there is clipping but that didn't work too well.  Now we de-weight the accelerometer that is clipping more (often one clips before the other).

With high vibes the copter doesn't usually flip and crash.  Normally long before that it's the altitude and position estimates that goes bad.

I'm open to trying to improve the vibration resistance more, we'd need to see logs of failure situations to see how it's failing and see if there are strategies that might work better.  For the moment, I think the best protection is for the GCSs to make the user aware with the data ardupilot is providing.

Jman841, 2016-04-11:
That's interesting. I have a few logs I can dig up from crashes that I believe were due to high vibrations, or it was a crash due to partial mechanical failure that caused high vibration, but, you can look to see the effects of these on the controller. I will send you them this week.

A little off topic, but, what is the major effects of low frequency vibrations? In the traditional heli side, the high frequency vibrations are less of an issue, however, a heli can exhibit quite strong low frequency vibrations if things are a little bit off. It does not seem to have the same detrimental effects as the high frequency vibrations from a multicopter, however, is there anything that a traditional heli user should be concerned with in regards to low frequency vibrations?

geofrancis, 2016-04-11:
i currently run a modified opentx firmware on my 9xr pro with openlrs for rc and telemetry. my handset can display error messages from the flight controller thats why i would like it as a message from flight controller rather than a function of the ground station.

R-Lefebvre, 2016-04-11:
Excessive low-freq vibrations, at least on a helicopter, can lead to basic flight control problems.  I think the issue is that the output from the PID's ends up saturating the servo outputs.  The output just becomes "mud" and the position the servos end up finding is something other than what is needed for stable flight.

I believe the same is true for large multirotors with limited control bandwidth due to rotor inertia.

This is why I pretty much don't even look at the VIBE log message data.  It is high-passed, operating with the assumption we are only concerned with the effect of accel aliasing into the sample rate which messes up the inertial solution. But it completely misses low frequency vibrations which hurt flight stability.

Naterater, 2019-01-07:
To me, this is less about vibrations causing issues with the position estimate, but instead it is about vibrations showing that something else is wrong (broken prop, motor bearing out, etc.). The failsafe should begin an automated RTL or other action to prevent more damage to the airframe or propulsion system components.

rmackay9, 2019-01-08:
@Naterater, the problem is that we lose altitude hold when vibration levels get really high so the RTL doesn't work.  What we need to do is reduce the EKF's reliance on the accels and increase reliance on the baro (or GPS)

Naterater, 2019-01-08:
Ok, but does it really have to be dynamic based on vibration levels? Shouldn't it be weighting on the baro fairly heavily anyway?

rmackay9, 2019-01-08:
@Naterater, yes, I think it needs to be dynamic.  The reason is that if we increase weighting on the baro then the altitude hold performance is worse.  I've got a really old video on my channel but can't find it immediately but in the video it shows a drop in altitude after fast forward flight which is caused by baro interference.  There's a newer video showing a bebop with ground effect compensation on/off which is also basically accomplished by **reducing** reliance on the baro.  So in short, we need a balance and/or dynamic changing of the weighting based on conditions.

rmackay9, 2019-10-05:
We have a duplicate issue here: https://github.com/ArduPilot/ardupilot/issues/3883.

rmackay9, 2020-01-27:
Copter-4.0 includes a vibration failsafe so I think I will close this issue.  If we need to re-open we can.

closed 2020-01-27 by rmackay9