ArduPilot/ardupilot #3884: streamRates values before REQUEST_DATA_STREAM rcvd

pwbecker, 2016-04-11:
Arducopter latest stable on APM 2.6

The initialization code for Arduplane has the comment   /* default stream rates to 1Hz */ and rates such as RAW_SENS etc are set to 1.  The corresponding code in Arducopter does not have this comment, and the default values are zero.

This causes a problem with minimOSD, which in the latest version does not send a REQUEST_DATA_STREAM.  Additionally, many configurations (like mine) have a telemetry radio, which means the Tx line from the minimOSD is not connected anyway (only one telemetry port on APM2 remember).

In the case when the telemetry radio connects, it sends a REQUEST_DATA_STREAM, and the minimOSD then works perfectly.  The problem is when the telemetry radio is not in use - that is when data is not updated on the OSD.

BTW as expected from the code, the same wiring set up works perfectly with Arduplane - OSD data is updated.

magicrub, 2016-04-11:
Latest stable does not work on APM 2.6.

pwbecker, 2016-04-11:
I meant latest stable that is APM2 compatible - its automatically selected based on the board connected at time of download.

magicrub, 2016-04-11:
APM 2.6 was phased out at different times for the different platforms.
Being specific would be appreciated, however, it probably won't help
because APM has been depreciated and no longer supported.

pwbecker, 2016-04-11:
V3.2.1. I updated the first post accordingly.

night-ghost, 2016-06-06:
Yes I confirm that this problem also present on recent releases. Moreover, there is no any MAVLINK_MSG_ID_STATUSTEXT messages before REQUEST_DATA_STREAM.

Tere are many devices on the weak processors who listen mavlink but did not request anything - eg OSD, beacons, emergency rescue systems, which are typically connected in parallel to telemetry modems. But if there is no telemetry ARDIPILOT needs an additional parameter, which shows that mavlink stream shall be sended in that port without request.

rmackay9, 2016-06-06:
You should be able to force copter to send data even without it receiving a specific request from the device by setting one of the SR0_* or SR1_* parameters to 1

night-ghost, 2016-06-06:
Yes this is correct for almost all messages - but not for MAVLINK_MSG_ID_STATUSTEXT! This message is not sent without REQUEST_DATA_STREAM at all. 

Old OSD code wasn't able to show messages so there wasn't a problem but new OSD code can! So everyone thinks that this is OSD issue where it simply don't receives that messages.

pwbecker, 2016-06-06:
@rmackay9 I tried setting SR* via APM Planner on Linux, and it did **not** take effect for the OSD.  When I did it via Android Tower, it did work.  Perhaps APM Planner restores SR*, or does not set the saved values?

estebanavv, 2016-06-06:
@pwbecker I think the parameters are properly set, but when you use Android Tower you're using radio telemetry and it is sending REQUEST_DATA_STREAM in consequence OSD works.

pwbecker, 2016-06-06:
@estebanavv sorry, perhaps I wasn't clear.  If I set SR* via APM Planner, OSD would not work after a reboot.  Any time I connected the telem radio, of course the stream started and the OSD worked.

When I used TOWER to set the SR* parameters, then the OSD worked, even after a reboot.

estebanavv, 2016-06-06:
@pwbecker and how do you connect Tower to avoid messing with the stream read by the OSD?
Thank you.

pwbecker, 2016-06-06:
I leave the tx line of the OSD disconnected, so that the OSD only receives data from the APM. Does that make sense?

estebanavv, 2016-06-06:
> LEGAL DISCLAIMER: The information contained in this message is intended
> solely for the individual or entity to which it is addressed and may
> contain confidential and/or privileged material.
> Any unauthorized dissemination, distribution, copying or taking action in
> reliance upon this information by persons or an entity other than the
> intended recipient is expressly prohibited. If you have received this email
> in error please notify the sender immediately and delete this message and
> any attachments.

Yes but when you do that the radio sends the request and the OSD receives
the feed.

http://www.rcgroups.com/forums/showpost.php?p=34952968&postcount=895

Not  that it helps, but it explains why you're getting messages in tower.

Best regards,
Esteban.

rmackay9, 2016-06-06:
Ah, ok.  so the issue is that text messages are only sent if we've detected a heartbeat on that channel.  We could change that so that it is based on either having received a heartbeat or having non-zero SR_ parameters.  I can't promise that I'll get to this soon but the issue is clear (to me anyway).

pwbecker, 2016-06-07:
@rmackay9 Thanks.  I think there may be two issues here - the one you have identified, and also something about using APM Planner to set SR parameters.  For me, I am sorted, as I modded the minimOSD code to send a REQUEST_DATA_STREAM, and also, once I had set the SR params using Tower (NOT using APM Planner), then it sent MAVLink even without the REQUEST_DATA.

@estebanavv  I explained exactly that in the original post.

marcmerlin, 2016-06-12:
Thanks. I had the same exact problem. I reported it here
http://discuss.ardupilot.org/t/i-dont-seem-to-get-mavlink-warning-messages-over-telem-port-in-many-cases/9498
but it seems no one saw it.

Indeed for now, I 'm forced to plug a 3DR radio in my OSD port to "prime" it after boot

magicrub, 2016-06-12:
Not sending the text messages to ports that have never seen data rx, such as a heartbeat,  was by design. Perhaps that design is flawed. I don't see any down side to just spamming all ports instead of just ports that have 2way traffic.

magicrub, 2016-06-12:
I broke this so I'll fix it.

marcmerlin, 2016-06-12:
@magicrub thanks. Note that even some android base stations are failing to receiving arming error messages now. Only one out of my 3 android base stations sends the right traffic to enable arming messages (and it's not tower, tower doesn't work).
Also, indeed with an OSD you want to receive data and the OSD may not send data back to you or even anything. People who don't want spam on their serial port can turn off the messages they don't need with SRX_Y (although to be fair they don't all seem very well documented as to which one sends which extra messages).

magicrub, 2016-06-12:
The sendtext spam is event driven, not a stream,  so SRx params do not
affect it.

marcmerlin, 2016-06-13:
Sorry, my bad. I guess if someone really cares, we can add an SRX option to turn them off.

magicrub, 2016-06-13:
Better solution is to send to all active steams plus ones that have non
zero SRx values.

magicrub, 2016-06-14:
This should now be fixed for all platforms: https://github.com/ArduPilot/ardupilot/commit/b4b0ec404c16c9cf9a9afec4e88092ce18aff956
Any port that streams anything (any SRx_ non-zero) then sendtexts are now also sent.

closed 2016-06-14 by magicrub