ArduPilot/ardupilot #3893: Plane: RPM over serial port

naktinis, 2016-04-12:
On our fixed-wing aircraft we are using UAV Factory's [Generator Power Unit](http://www.uavfactory.com/product/9) which provides RPM among other variables (load current, battery voltage etc.) as serial data.

It would be great if we could connect this serial port directly to the autopilot and ArduPlane would update internal RPM value from these messages.

Would this functionality be useful for other users? What is the best approach to adding this functionality? From what I see, it should be possible to add a new `SERIALX_PROTOCOL` and a "driver" `RPM_UAVFactoryGPU` that could parse the specific message (just like there are "drivers" for each range finder type, `AP_RangeFinder_LightWareSerial` etc.).

R-Lefebvre, 2016-04-12:
I definitely think that something like this would be very useful. I had started an I2C protocol for getting data from an engine management system a year ago, but it died on the vine.

You could add a serial interface for that specific system.  However, I would prefer if somebody made a generic engine management interface operating over UAVCan.  It would require an adaptor between the Ardupilot avionics and the UAV Factory system in this case.  But hopefully in future, 3rd party vendors would adopt the standard.

CANbus is definitely the direction we want to be pushing in the future for these systems.

magicrub, 2016-04-13:
I agree that going CAN is the way to go but that is not an option for this particular hardware. Yes @naktinis the correct implementation is to add a new SERIALx_PROTOCOL. Pull requests welcome!

jpkh, 2016-04-18:
There are people asking similar feature but with PWM signaling. Maybe a Serial input would be best and then we can have multiple motor input coming from external board like IOBoard or Teensy based boards. 

Hobbywing have now ESCs that provides RPM output and I bet more will come...

http://discuss.ardupilot.org/t/how-to-read-rpm-signal-from-esc/8326/7

Naterater, 2019-01-07:
The BLHELI improvements have recently made this possible (but probably for only  BLHELI telemetry protocol). I think an adaption of that work might make this possible without too much effort.

auturgy, 2019-01-07:
@naterator the blheli32 telemetry isn’t relevant here.
This is basically a request for a new serial driver.

Naterater, 2019-01-07:
This shouldn't be plane-specific then, is that right?

auturgy, 2019-01-07:
The serial driver wouldn’t be, but the equipment in this case is. Probably overtaken by events so doesn’t really matter: I was just clarifying that blheli telemetry is irrelevant here.

peterbarker, 2026-11-15:
I'm guessing the need for this is long, long gone.  Modern ArduPilot would probably use a script to read this and feed data through to the relevant places.

Closing this one now.

closed 2026-11-15 by peterbarker