ArduPilot/ardupilot #3894: Tools: install-prereqs-ubuntu.sh  script fails , needs updating

AndKe, 2016-04-12:
Tested on fresh install of Ubuntu 16.04
First : python-wxgtk2.8  is no longer available,  python-wxgtk3.0  should be  used.
Then the script command "sudo pip2 -q install $PYTHON_PKGS"   fail with :
"Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-o9DnpF/pymavlink/"

if the script got past that, next issues are:
Add /opt/gcc-arm-none-eabi-4_9-2015q3/bin to your PATH [Y/n]?Y
./install-prereqs-ubuntu.sh: line 85: grep: command not found
Add /home/andre/prog/ardupilot/Tools/scripts/ardupilot/Tools/autotest to your PATH [Y/n]?Y
./install-prereqs-ubuntu.sh: line 94: apt-cache: command not found

magicrub, 2016-04-13:
> python-wxgtk2.8 is no longer available, python-wxgtk3.0 should be used

Is this a Python 2.x to 3.x change? That sounds like a significant change.

AndKe, 2016-04-13:
no, nothing that drastic,  the default python is still 2.7.11  - just some upgraded libraries.

lucasdemarchi, 2016-04-15:
@AndKe and you don't have grep in Ubuntu 16.04 even after installing build-essentials? Or is it because the apt-get call failed and then things weren't installed... ?

AndKe, 2016-04-16:
@lucasdemarchi of course there is grep , it's used in many of my scripts all the time, - I assume it's something trivial with the syntax or something - that I can figure out in a spare moment..

lucasdemarchi, 2016-04-16:
./install-prereqs-ubuntu.sh: line 85: grep: command not found

AndKe, 2016-04-16:
@lucasdemarchi - yes, I do still see it, - let's forget the trivial stuff, (if that's a problem om more than one test computer, I'll fix it)
Now, if we can focus on a solution for the library-issue.  :)

lucasdemarchi, 2016-04-16:
The thing is that I think the apt-get failed and then things that are needed weren't installed.

AndKe, 2016-04-16:
sure it failed :)   - because python-wxgtk2.8 is not in 16.04 ...  :)   so  - what can we do about that ?

lucasdemarchi, 2016-04-16:
Probably the same we did in CI scripts to differentiate 14.04 and 12.04: we parse the distro name from /etc/os-release and install the different SW based on that. If mavproxy (the one that actually use python-wxgtk) works with both versions, then we are done. Otherwise we would need some patches there, too.

lucasdemarchi, 2016-04-16:
I can take a look into this on Monday if nobody implements it before.

AndKe, 2016-04-17:
do we know all parts that need python-wxgtk2.8 , and whatever they do work with 3.0 ? (I am familiar with python dependencies & backwards compatibility)

AndKe, 2016-04-17:
I can confirm that mavproxy (and --console) works fine with python-wxgtk3.0  
- will make a PR with the fix tomorrow.

AndKe, 2016-04-17:
here: https://github.com/ArduPilot/ardupilot/pull/3917

peterbarker, 2017-12-19:
I think we must be all good for this by now :-)

closed 2017-12-19 by peterbarker