ArduPilot/ardupilot #3895: Feature Request, Rallypoint Pass through to Home

Jman841, 2016-04-12:
Currently from my understanding if I set a rallypoint with Arduplane the plane will hit RTL then go to this point and loiter there until I give it another command. I would like the desired action for the plane to go first to the nearest rallypoint then continue to Home. This is desired just in case for some reason I do not have telemetry link at the desired rallypoint. or for some reason I need to use the RC Controller but the rally point is at a point that is beyond line of sight.

Platform: Copter and Plane

Thank you,

Jason

iskess, 2016-04-12:
This would also help when flying alongside a tall mountain ridge. An RTL to home could cause the plane to turn into the cliff wall. Rally Points could be used to assure that with a radio failsafe at any point, the closest heading to is always away from the cliff. 
This request would allow the plane to then return home after the rally. 
Actually this is still just a work around. A better solution to my case would be general Cardinal direction that could be entered to control the direction of the initial turn of an RTL.

iskess, 2016-04-12:
Maybe we just need to incorporate DO_LAND_START to work with a Rally point so the plane will head to the approach sequence which is at home.

Jman841, 2016-04-12:
I don't want it to start the landing sequence right away. I always set RTL before beginning the landing to ensure everyone is ready and the runway is clear. Simply for it to use the rally point as a pass through in case there is an obstacle where a direct line back to RTL is not a good option, then head to RTL. The only thing I would like to see is the set altitude for the rally point to be maintained all the way through to RTL.

IamPete1, 2021-07-12:
I think the existing DO_LAND_START functionality would allow this?

peterbarker, 2021-07-30:
@IamPete1 pretty sure.

Closing this.

closed 2021-07-30 by peterbarker