ArduPilot/ardupilot #3903: Req: CAMERA_FEEDBACK should contain shutter count

AndKe, 2016-04-15:
Shutter feedback should be counted and a part of this message.
This way operator could easily see not only how many times camera was expected to shoot (img_idx) , but also how many time the shutter actually operated.

[image]

WickedShell, 2017-02-10:
Agreed, this is a very good canidate for MAVLink 2 extensions. This is actually on my radar, and will probably be what forces me to implement Java MAVLink2 support.

closed 2018-03-13 by OXINARF

AndKe, 2018-03-20:
compiled master today. 
All I see in mavproxy is:
CAMERA_FEEDBACK.alt_msl        CAMERA_FEEDBACK.foc_len        CAMERA_FEEDBACK.pitch          CAMERA_FEEDBACK.yaw
CAMERA_FEEDBACK.alt_rel        CAMERA_FEEDBACK.img_idx        CAMERA_FEEDBACK.roll           
CAMERA_FEEDBACK.cam_idx        CAMERA_FEEDBACK.lat            CAMERA_FEEDBACK.target_system  
CAMERA_FEEDBACK.flags          CAMERA_FEEDBACK.lng            CAMERA_FEEDBACK.time_usec 

why ?

OXINARF, 2018-03-20:
@AndKe Probably because you didn't enable MAVLink2.

AndKe, 2018-03-20:
@OXINARF 
I connected to current AC master in SITL: - (all links set to Mavlink2 & restarted)  (then let QGC connect using UDP to run the mavlink inspector for graphic presentation:)
[image]

OXINARF, 2018-03-20:
It's quite likely that QGC doesn't support the MAVLink extension yet.

AndKe, 2018-03-20:
QGC got it's submodules updated recently, but more importantly:

I've also build mavproxy from master, and:  (it's not even an option)  - actually, that's where I tried first.

```
STABILIZE> graph CAMERA_FEEDBACK.
CAMERA_FEEDBACK.alt_msl        CAMERA_FEEDBACK.flags          CAMERA_FEEDBACK.lat            CAMERA_FEEDBACK.roll           CAMERA_FEEDBACK.yaw            
CAMERA_FEEDBACK.alt_rel        CAMERA_FEEDBACK.foc_len        CAMERA_FEEDBACK.lng            CAMERA_FEEDBACK.target_system  
CAMERA_FEEDBACK.cam_idx        CAMERA_FEEDBACK.img_idx        CAMERA_FEEDBACK.pitch          CAMERA_FEEDBACK.time_usec 
```     
Do we agree that mavproxy has updated mavlink2 ?

WickedShell, 2018-03-20:
@AndKe I haven't upstreamed the extensions to the mavlink/mavlink repo yet. At the moment they are living in https://github.com/ArduPilot/mavlink and QGC doesn't point at that mavlink fork. I can't speak to where MAVProxy/pymavlink are drawing the protocol files from...

AndKe, 2018-03-20:
Ah, thanks - in that case, I should be more patient :)

OXINARF, 2018-03-21:
MAVProxy uses pymavlink and, AFAIK, there hasn't been a pymavlink release that includes the updated message. You can install a development version of pymavlink and then try again with MAVProxy (make sure it is using MAVLink2, I'm not sure if you need to pass `--mav20` or not).

commit 0e6213a4 (2018-03-13): Camera: Track number of completed events

Closes #3903