ArduPilot/ardupilot #3912: clone in GitHub doesn't work

px4stm32, 2016-04-16:
Hello,

I try to open https://github.com/ArduPilot/ardupilot with GitHub in Windows 7, but it doesn't work.
I use this outdated tutorial: http://ardupilot.org/dev/docs/where-to-get-the-code.html

When I click on the Button "...use it in GitHub Desktop", GitHub opens but I'm not able to clone it.
The error log shows this:

`2016-04-16 20:20:03.1792|INFO|thread:16|DashboardViewModel|Took 2ms to Create RepositoryViewModel for location C:\Users\Administrator.laptop-PC\Documents\GitHub\ardupilot
2016-04-16 20:17:28.4704|INFO|thread: 1|GitHubEnvironment|ACTIVATION DATA RAW
2016-04-16 20:17:28.4860|INFO|thread: 1|GitHubEnvironment|[http://github-windows.s3.amazonaws.com/GitHub.application]: '0'
2016-04-16 20:17:28.4860|INFO|thread: 1|GitHubEnvironment|ACTIVATION DATA UNENCODED
2016-04-16 20:17:28.4860|INFO|thread: 1|GitHubEnvironment|[http://github-windows.s3.amazonaws.com/GitHub.application]: '0'
2016-04-16 20:17:28.4860|INFO|thread: 1|CommandLineHandler|Parsing command line arguments:http://github-windows.s3.amazonaws.com/GitHub.application
2016-04-16 20:17:28.5328|INFO|thread: 1|CommandLineHandler|Unprocessed args: http://github-windows.s3.amazonaws.com/GitHub.application
2016-04-16 20:17:28.5484|WARN|thread: 1|CommandHandler|Exception occurred while opening path 'http://github-windows.s3.amazonaws.com/GitHub.application'
System.NotSupportedException: The given path's format is not supported.
   at System.Security.Util.StringExpressionSet.CanonicalizePath(String path, Boolean needFullPath)
   at System.Security.Util.StringExpressionSet.CreateListFromExpressions(String[] str, Boolean needFullPath)
   at System.Security.Permissions.FileIOPermission.AddPathList(FileIOPermissionAccess access, AccessControlActions control, String[] pathListOrig, Boolean checkForDuplicates, Boolean needFullPath, Boolean copyPathList)
   at System.Security.Permissions.FileIOPermission..ctor(FileIOPermissionAccess access, String[] pathList, Boolean checkForDuplicates, Boolean needFullPath)
   at System.IO.Path.GetFullPath(String path)
   at GitHub.Helpers.CommandHandler.SelectRepositoryByPath(String path)
2016-04-16 20:17:28.5952|INFO|thread: 1|AppInstance|Starting up as master instance of GitHub Desktop
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| #########################################
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| GitHub Desktop started. VERSION: 3.0.17.0
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| Build version: 6ea318260af5bc390b90474e5adde8c44306b23f
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***************************************
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***                                 ***
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***                                 ***
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***        Have a problem?          ***
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***    Email support@github.com     ***
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***      and include this file      ***
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***                                 ***
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***                                 ***
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| ***************************************
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| OS Version: Windows 7 Service Pack 1 6.1.7601.65536 amd64
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| CLR Version: 4.0.30319.17929
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| Current culture: de-DE
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| Environment.CurrentDirectory: C:\Users\Administrator.laptop-PC\AppData\Local\Apps\2.0\5G8TY7EB.D0N\HW0D1EQT.B30\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| currentProcess.StartInfo.WorkingDirectory: 
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| Terminal Services session: no
2016-04-16 20:17:28.6264|INFO|thread: 5|StartupLogging| Location: C:\Users\Administrator.laptop-PC\AppData\Local\Apps\2.0\5G8TY7EB.D0N\HW0D1EQT.B30\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354\GitHub.exe
2016-04-16 20:17:28.6576|INFO|thread: 5|StartupLogging| ActivationUri: https://github-windows.s3.amazonaws.com/GitHub.application
2016-04-16 20:17:28.6576|INFO|thread: 5|StartupLogging| System.Environment.CommandLine: C:\Users\Administrator.laptop-PC\AppData\Local\Apps\2.0\5G8TY7EB.D0N\HW0D1EQT.B30\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354\GitHub.exe
2016-04-16 20:17:28.8448|INFO|thread: 5|GitEnvironment|Process set up with this SSH Agent info: (null):(null)
2016-04-16 20:17:28.8448|INFO|thread: 5|GitEnvironment|PATH is C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\cmd;C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\usr\bin;C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\usr\share\git-tfs;C:\Users\Administrator.laptop-PC\AppData\Local\Apps\2.0\5G8TY7EB.D0N\HW0D1EQT.B30\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354;C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\lfs-amd64_1.1.0;C:\Windows\Microsoft.NET\Framework\v4.0.30319;C:\Program Files (x86)\Mozilla Firefox;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin
2016-04-16 20:17:28.8448|INFO|thread: 5|StartupLogging| =====================================================
2016-04-16 20:17:28.8448|INFO|thread: 5|StartupLogging|  DIAGNOSTICS                                        |
2016-04-16 20:17:28.8448|INFO|thread: 5|StartupLogging| =====================================================
2016-04-16 20:17:28.8448|INFO|thread: 5|StartupLogging| Git Extracted: 'False:
2016-04-16 20:17:28.8448|INFO|thread: 5|StartupLogging| PortableGit Dir Exists: 'C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0'
2016-04-16 20:17:28.8604|ERROR|thread: 5|PortableGitManager|git.exe doesn't exist at 'C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\cmd\git.exe'
2016-04-16 20:17:28.8760|ERROR|thread: 5|StartupLogging| MISSING Git Executable: 'C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\cmd\git.exe'
2016-04-16 20:17:28.8760|ERROR|thread: 5|StartupLogging| MISSING PATH!!: 'C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\cmd'
2016-04-16 20:17:28.8760|ERROR|thread: 5|StartupLogging| MISSING PATH!!: 'C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\lfs-amd64_1.1.0'
2016-04-16 20:17:28.8760|INFO|thread: 5|StartupLogging| ----------------------------------------------------
2016-04-16 20:17:28.8760|INFO|thread: 5|StartupLogging| PATH: C:\Program Files (x86)\Mozilla Firefox;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin
2016-04-16 20:17:28.9696|INFO|thread: 5|StartupLogger|Proxy information: (None)
2016-04-16 20:17:28.9696|INFO|thread: 5|StartupLogger|Couldn't fetch creds for proxy
2016-04-16 20:17:29.1880|INFO|thread: 5|UrlProtocolInstaller|Registered UrlAppLauncher protocol: "C:\Users\Administrator.laptop-PC\AppData\Local\Apps\2.0\5G8TY7EB.D0N\HW0D1EQT.B30\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354\GitHub.exe" -u="%1"
2016-04-16 20:17:29.2192|INFO|thread: 1|HardwareRenderingHelper|Making a WMI query to discover any video PNPID entries
2016-04-16 20:17:29.2348|INFO|thread: 5|ShortcutManager|Shortcut 'C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\GitHub.appref-ms' missing. Assuming a first run.
2016-04-16 20:17:29.2348|INFO|thread: 5|ShortcutManager|Copying the Start Menu shortcut 'C:\Users\Administrator.laptop-PC\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\GitHub, Inc\GitHub.appref-ms' to 'C:\Users\Administrator.laptop-PC\AppData\Local\GitHub\GitHub.appref-ms'
[... cut ...]
2016-04-16 20:20:00.6230|WARN|thread: 1|StandardUserErrors|Showing user error Please check your log file for more details, or email support if you are still having problems.
GitHub.IO.ProcessException: 20:20:00.045059 git.c:344               trace: built-in: git 'clone' '--progress' '--recursive' '--' 'https://github.com/px4stm32/ardupilot.git' 'ardupilot'
20:20:00.177066 run-command.c:334       trace: run_command: 'git-remote-https' 'origin' 'https://github.com/px4stm32/ardupilot.git'
Cloning into 'ardupilot'...
`

OXINARF, 2016-04-16:
That seems like an issue with the GitHub for Windows client, not an issue with our repo. Have you tried to reinstall the client?

px4stm32, 2016-04-16:
I reinstalled Windows 7 now and the clone failed again:

2016-04-16 23:08:59.1249|INFO|thread:32|AsyncReaderWriterLock|Released concurrent scheduler for operation: 15 at RepositoryViewModel.cs:318 after blocking 0.0215339s
2016-04-16 23:04:38.3572|INFO|thread: 1|GitHubEnvironment|ACTIVATION DATA RAW
2016-04-16 23:04:38.3884|INFO|thread: 1|GitHubEnvironment|[base64:LXU9Z2l0aHViLXdpbmRvd3M6Ly9vcGVuUmVwby9odHRwczovL2dpdGh1Yi5jb20vQXJkdVBpbG90L2FyZHVwaWxvdA==,L3Jlc3RhcnRlZA==,LS1jZD1DOlxQcm9ncmFtIEZpbGVzICh4ODYpXEdvb2dsZVxDaHJvbWVcQXBwbGljYXRpb25cNTAuMC4yNjYxLjc1]: '0'
2016-04-16 23:04:38.3884|INFO|thread: 1|GitHubEnvironment|ACTIVATION DATA UNENCODED
2016-04-16 23:04:38.3884|INFO|thread: 1|GitHubEnvironment|[-u=github-windows://openRepo/https://github.com/ArduPilot/ardupilot]: '0'
[/restarted]: '1'
[--cd=C:\Program Files (x86)\Google\Chrome\Application\50.0.2661.75]: '2'
2016-04-16 23:04:38.3884|INFO|thread: 1|CommandLineHandler|Parsing command line arguments:-u=github-windows://openRepo/https://github.com/ArduPilot/ardupilot
/restarted
--cd=C:\Program Files (x86)\Google\Chrome\Application\50.0.2661.75
2016-04-16 23:04:38.4664|INFO|thread: 1|CommandLineHandler|Unprocessed args: 
2016-04-16 23:04:38.4976|INFO|thread: 1|AppInstance|Starting up as master instance of GitHub Desktop
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| #########################################
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| GitHub Desktop started. VERSION: 3.0.17.0
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| Build version: 6ea318260af5bc390b90474e5adde8c44306b23f
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***************************************
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***                                 ***
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***                                 ***
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***        Have a problem?          ***
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***    Email support@github.com     ***
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***      and include this file      ***
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***                                 ***
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***                                 ***
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| ***************************************
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| OS Version: Windows 7 6.1.7600.0 amd64
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| CLR Version: 4.0.30319.17929
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| Current culture: de-DE
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| Environment.CurrentDirectory: C:\Users\test\AppData\Local\Apps\2.0\RA6C9N6Q.6VX\5XZTA9NZ.55Q\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354
2016-04-16 23:04:38.5288|INFO|thread: 4|StartupLogging| currentProcess.StartInfo.WorkingDirectory: 
2016-04-16 23:04:38.5444|INFO|thread: 4|StartupLogging| Terminal Services session: no
2016-04-16 23:04:38.5444|INFO|thread: 4|StartupLogging| Location: C:\Users\test\AppData\Local\Apps\2.0\RA6C9N6Q.6VX\5XZTA9NZ.55Q\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354\GitHub.exe
2016-04-16 23:04:38.5756|INFO|thread: 4|StartupLogging| ActivationUri: https://github-windows.s3.amazonaws.com/GitHub.application
2016-04-16 23:04:38.5756|INFO|thread: 4|StartupLogging| System.Environment.CommandLine: C:\Users\test\AppData\Local\Apps\2.0\RA6C9N6Q.6VX\5XZTA9NZ.55Q\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354\GitHub.exe
2016-04-16 23:04:38.6536|INFO|thread: 1|HardwareRenderingHelper|Your video card appears to support hardware rendering. If this isn't the case and you see glitches
2016-04-16 23:04:38.6536|INFO|thread: 1|HardwareRenderingHelper|set the GH_FORCE_SW_RENDERING environment variable to 1
2016-04-16 23:04:38.6536|INFO|thread: 4|GitEnvironment|Process set up with this SSH Agent info: 1736:/tmp/ssh-VLHZeDzolwxa/agent.640
2016-04-16 23:04:38.6536|INFO|thread: 4|GitEnvironment|PATH is C:\Users\test\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\cmd;C:\Users\test\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\usr\bin;C:\Users\test\AppData\Local\GitHub\PortableGit_25d850739bc178b2eb13c3e2a9faafea2f9143c0\usr\share\git-tfs;C:\Users\test\AppData\Local\Apps\2.0\RA6C9N6Q.6VX\5XZTA9NZ.55Q\gith..tion_317444273a93ac29_0003.0000_c74cce3a838f9354;C:\Users\test\AppData\Local\GitHub\lfs-amd64_1.1.0;C:\Windows\Microsoft.NET\Framework\v4.0.30319;C:\Program Files (x86)\Google\Chrome\Application;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\
2016-04-16 23:04:38.6536|INFO|thread: 4|StartupLogging| =====================================================
2016-04-16 23:04:38.6536|INFO|thread: 4|StartupLogging|  DIAGNOSTICS                                        |
2016-04-16 23:04:38.6536|INFO|thread: 4|StartupLogging| =====================================================
2016-04-16 23:04:38.6536|INFO|thread: 1|GitHubEnvironment|ACTIVATION DATA RAW
2016-04-16 23:04:38.6536|INFO|thread: 1|GitHubEnvironment|[base64:LXU9Z2l0aHViLXdpbmRvd3M6Ly9vcGVuUmVwby9odHRwczovL2dpdGh1Yi5jb20vQXJkdVBpbG90L2FyZHVwaWxvdA==,L3Jlc3RhcnRlZA==,LS1jZD1DOlxQcm9ncmFtIEZpbGVzICh4ODYpXEdvb2dsZVxDaHJvbWVcQXBwbGljYXRpb25cNTAuMC4yNjYxLjc1]: '0'
[... cut ...]
2016-04-16 23:08:59.0779|INFO|thread: 1|AsyncReaderWriterLock|Acquiring concurrent scheduler for operation 15 at RepositoryViewModel.cs:318
2016-04-16 23:08:59.0919|INFO|thread:11|AsyncReaderWriterLock|Acquired concurrent scheduler for operation: 14 at RepositoryViewModel.cs:318 after waiting 0.0137192s
2016-04-16 23:08:59.0919|INFO|thread:11|AsyncReaderWriterLock|Released concurrent scheduler for operation: 14 at RepositoryViewModel.cs:318 after blocking 2.05E-05s
2016-04-16 23:08:59.0919|INFO|thread:11|AsyncReaderWriterLock|Acquired concurrent scheduler for operation: 15 at RepositoryViewModel.cs:318 after waiting 0.0247082s

OXINARF, 2016-04-16:
Looking at the log the issue is that at one point the connection the Github failed so it didn't complete the cloning.
Anyway, this isn't an issue with Ardupilot so I'll have to close it now. If you still have issues you should contact Github or open a post at discuss.ardupilot.org

closed 2016-04-16 by OXINARF