ArduPilot/ardupilot #3918: Parachute should turn off the motor only and not disarm the AP

Elios007, 2016-04-17:
#### Issue details
it has been 2 years we land only using parachutes in our fixed wings plane, we were and still using the manual parachute trigger. We were happy that this trigger is via Pixhawk lately but we found due to our 100 flights and parachute landing that the Pixhawk should not disarm the AP but only turn off the motor. Because today we have triggered the parachute, the door has opened but the parachute didn't pull out for a certain reason. If we were using the parachute via pixhawk, we would have crashed the plane, the best scenario is to make the parachute function to turn off the motor only and if for a reason the parachute didn't pull out, the pilot will have a last chance to switch the FBWA and land the plane on its belly.

#### Version
_What version was the issue encountered with_

#### Platform
arduplane

#### Airframe type
fixed wings plane

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-04-17:
I think this is a duplicate.

Elios007, 2016-04-18:
HI, 

What do you mean it is duplicate? 
if you mean my previous issue about the pre-check, it is not, as it was about the pre-check only. But if someone else has talked about it, would you please post the link issue here, as I tried to find it but I didn't.

Thanks for your help and time.

tridge, 2016-04-18:
It looks like you are asking for the current behaviour. The parachute code in APM:Plane only disables the motor in auto-throttle modes (such as AUTO). In manual throttle modes (such as FBWA) the motor remains active.
This was done for two reasons:

- so that if the parachute open fails the pilot has a chance to recover by switching to a manual throttle mode
- so that the pilot can use a manual throttle mode to guide the descent of the parachute

The control surfaces are not disabled when the parachute is released.

closed 2016-04-18 by tridge

Elios007, 2016-04-18:
Thanks for the clarification.