ArduPilot/ardupilot #3919: wp_negative is giving radius 30 meter regardless the input number.

Elios007, 2016-04-17:
#### Issue details
we were doing loiter today but we found that when wp_loiter_rad is positive radius is correct but when negative for a counter clockwise loiter, the radius is very small (30m) regardless the input number. We tried even -5000.

#### Version
3.5.2

#### Platform
- [ ] Plane

#### Airframe type
Fixed wings plane

#### Hardware type
Pixhawk

#### Logs
https://www.dropbox.com/sh/hkt7zy6iuu8dtb2/AADARqvwErd4tKfHBBRAO74Ka?dl=0

magicrub, 2016-04-17:
I believe this was fixed by Michael dB just after the v3.5.2 release, will
check and follow up.

https://www.dropbox.com/sh/hkt7zy6iuu8dtb2/AADARqvwErd4tKfHBBRAO74Ka?dl=0

WickedShell, 2016-04-18:
Tridge found and fixed it with 62574b35b52ce65a. Was a misplaced paren pair on the abs() in navigation.cpp relates to the loiter changes from late Feb.

closed 2016-04-18 by WickedShell