ArduPilot/ardupilot #3920: Copter: Throw mode should (optionally) switch to AUTO after completion

rmackay9, 2016-04-18:
#### Issue details
dogintwater user requested throw mode switch to AUTO after it completes.  We should perhaps add a THROW_NEXT_MODE (or similarly named) parameter to allow the user to specify what mode it should switch to after the throw is complete.  Valid modes would certainly include Guided and Auto.  Other modes are questionable including Loiter, Stabilize, etc because the user would need to get to the transmitter sticks quite quickly.

#### Version
Copter-3.4

ferret-guy, 2016-06-01:
I would love this, possibly instead add it as the first wp in an auto mission

rmackay9, 2016-08-02:
This is included in master now and will go out with Copter-3.4-rc2

closed 2016-08-02 by rmackay9