ArduPilot/ardupilot #3921: Setting guided points are always reported as accepted even if the result was rejected

WickedShell, 2016-04-18:
#### Issue details
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L644 will always be reported as accepted, even if it was rejected due to the current mode (or other problems).

#### Version
Master

#### Platform
- [x] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover

nitirohilla, 2016-04-22:
@WickedShell  As per my understanding, guided points should be rejected when:

1. Current mode is not GUIDED mode, for all vehicles.
2. In case of Copter, when cmd.id is not equal to "MAV_CMD_NAV_WAYPOINT" or "MAV_CMD_CONDITION_YAW".

Can you please help me understand the other problems in which guided points should be rejected.

WickedShell, 2016-04-22:
Right, at the moment we report that we accepted the point even if the
vehicle rejects. IE internally we rejected the guided point (we weren't in
GUIDED, or something else was wrong with it), but we told the GCS that we
accepted it. This isn't a functional change for the autopilot side, its
just to enable better feedback on ground stations.

WickedShell, 2016-04-30:
Addressed in #3985

closed 2016-04-30 by WickedShell

nitirohilla, 2016-05-02:
Thanks @WickedShell