ArduPilot/ardupilot #3924: "Point Camera here" issues with servo-based gimbal on Plane 3.5 and Mission Planner

alexchernoff, 2016-04-18:
#### Issue details
Dear all,

I have a servo based gimbal on a fixed wing. Set up the angles and servo values as per the "Camer gimbal with servos" docs and I can control camera pan/tilt from the RC. 

When I use Mission Planner to "Point camera here", it rotates towards the "target", and that's it. It doesn't adjust the camera position, there is no more manual control, trying to "point camera" to another location does nothing. In MP there are some commands such as "RC targeting", "neutral", "GPS targeting" etc. those do nothing.

Is there perhaps a better document on how this feature works, and how it is supported by MP?

thanks!

#### Version
3.5.2

#### Platform
PLANE 

#### Airframe type
Fixed wing

#### Hardware type
pixhawk

#### Logs
N/A

amilcarlucas, 2017-09-25:
Does it correctly stabilize the camera when you move the plane around. Only when it does is the gimbal correctly configured.

Can you please retest this with Arduplane 3.8 and the latest mission planner ?

Naterater, 2019-01-04:
Is this still an issue?

anantshukla, 2019-01-24:
Yes

> Is this still an issue?

Yes. When I click on Point camera here, only the pitch changes. The camera doesn't roll. I've got a 2 axis gimbal.
Also, do you have an idea how I could point the camera using a dronekit script.

amilcarlucas, 2019-01-24:
Use DO_SET_ROI commands.

gediminasgu, 2019-01-28:
ArduPilot can point camera only on Tilt and Pan axis. If you have Tilt and Roll it won't work. It's just not implemented.
https://github.com/ArduPilot/ardupilot/blob/3a53c5f99c08136ad19498ee443caba684841131/libraries/AP_Mount/AP_Mount_Backend.cpp#L140

IamPete1, 2021-09-11:
I'm going to close this, it seems that we just don't support a tilt and roll. If there is some interest sin that feature please open a new issue

closed 2021-09-11 by IamPete1