ArduPilot/ardupilot #3925: Plane V Tail code should be used for all aircraft with a real rudder and elevator (vtail or straight configuration) and mix based on plane roll angle

marcmerlin, 2016-04-18:
To be clear, this should exclude flying wings which are a different animal as they do not have a real rudder to start with.

For  regular planes and Vtail planes, autopilot rudder and elevator output should be mixed into the elevator and rudder channels depending on whether it's Vtail (add a 45 degree offset) and according to the plane roll angle.
What I mean here is that if a plane with a straight elevator/rudder is banked at 45 degrees, you would send the same output to rudder/elevator as you would to a Vtail plane flying straight (ok, there may be a small offset due to the rudder and elevator control surfaces being of different size on a normal plane).
But the point remains, the vtail code ought to work on all planes, you then compute the mix based on the roll angle, and then you knife edge and optimized control surface use regardless of roll angle.
This would also avoid things like my crash in this issue: https://github.com/ArduPilot/ardupilot/issues/3875

marcmerlin, 2016-04-26:
@magicrub  do know the plane code and could you validate whether this is a reasonable idea to implement?

magicrub, 2016-04-26:
I think in theory it sounds reasonably but the devil is in the details on implementing it within the controller.

marcmerlin, 2016-04-27:
Thanks @magicrub . For my own understanding, do you know who else works on the plane code besides @tridge ?
I realize this is a bigger fix than the last bug I found, but at the same time I'll now feel uncomfortable flying AP manually knowing that I can't trust it to recover from upset conditions.

magicrub, 2016-04-27:
@WickedShell is another plane power user developer. Sorry that you've lost trust in arduplane. Seems like a corner case to me, switching to auto/RTL when inverted. Perhaps that's an easier short term solution that checks for inversion when switching to auto and does a controlled roll to recover?
Auto can fly up side down but it just needs to know about.

marcmerlin, 2016-04-27:
@magicrub thanks confirming, I thought @WickedShell was a plane dev too, but wasn't sure.
As for losing trust, well, I know AP will fly ok if it's in control the entire time, but I do often fly manu with a switch to flick if things go wrong, and now I know I can't do this and expect things will go well, even if I have 100m altitude in the bank.
On the plus side, I rarely lose control like this, it's usually when I'm flying visually from the ground (not FPV), and I'm a bit at the edge of what I can see well. Obviously for a night flight, flying from the ground gets a lot harder.
But back to AP, I now know that it does pretty bad things in the 80 to 100 degree roll range, and given that, it makes me pretty uneasy to fly my fairly precious plane knowing the problem is still there (took me 4 days of solid work to fix again after that crash, and it's a small miracle that the plane's body did not explode in pieces on impact)

magicrub, 2016-04-27:
Your aircraft should not be in the 80 to 100 degree zone, the autopilot
should limit your bank. Have you considered flying in FBW?

WickedShell, 2016-04-27:
As a reference it does bad things in the excess of 85 degrees pitch either way. (the control loops exhibit gimbal lock, even though the EKF/AHRS isn't suffering from it). This is somewhat demonstrable by watching what the HUD as you tilt the autopilot towards those pitch limits.

marcmerlin, 2016-04-27:
@magicrub: Sure I could fly FBWA, I do for FPV flying, but for takeoff/landing, I use MANUAL. Half because I'm an RC pilot used to the normal RC way of flying, but also because for landing I have crow flaps mixed by my transmitter but that won't work through any AP controlled mode (setting flaperons + flaps at the same time, which is what I have, I don't think is possible in AP).
@WickedShell: agreed on the 10-20 death corner for roll (around 90 degrees). Yes, of course, you can do your best not to get there, but at the same time, I've always considered AP as my 'oh shit' option (not unlike the similarly named button in a cirrus) that you press when you've lost it and you'd like the computer to bring back to level flight :)  I virtually never use it, but I'd like to know it's there and that I can depend on it.

magicrub, 2017-06-06:
I think this is implemented and can be closed. There has always been mixing like this but now there are limits on V3.6 on elevator when exceeding the max roll limit.

Naterater, 2018-12-01:
I can't comment on if this is implemented or not, but very related is the aileron behavior at extreme pitch values in FBWA. If the plane is expected to pitch up 90-degrees to climb, the angle may overshoot just a little. At this overshoot pitch point, the plane assumes the pitch to be decreasing, but heading in the opposite direction. At that moment, the plane rolls 180-degrees in an attempt to keep from "going inverted." at extreme pitch values, the ailerons need to be mapped to yaw. At extreme bank values, the rudder needs to be mapped to pitch. 

I would also like to see this implemented, but for all scenarios. I think it would help TECS a lot in turns as well, and it could help out the sideslip controller too.

Naterater, 2019-01-07:
@magicrub, I think this should probably be closed. I've contemplated this a bit more, and the rudder controls should always be done in the vehicle-frame, not earth-frame in order to coordinate flight controls no matter the attitude. I've had very good luck tuning vehicles at high bank angles (60+ degrees of bank on turns).

IamPete1, 2021-07-25:
@magicrub and @Naterater  seem to think this can be closed, so I'm going to do that.

closed 2021-07-25 by IamPete1