ArduPilot/ardupilot #3929: Common: Implement Follow-me by Mavlink msg

lvale, 2016-04-19:
Now that the accorded Mavlink message has gone into Mavlink, we should think on implementation on Copter/Plane/Rover

https://github.com/mavlink/mavlink/pull/526/files

lvale, 2018-08-19:
Having this @rmackay9 would be a solution for the users that were using the broken Tower solution

lvale, 2018-08-20:
@peterbarker would enable a free alternative solution to use follow-me on mobile devices

peterbarker, 2018-08-21:
On Mon, 20 Aug 2018, Luis Vale Gonçalves wrote:

> @peterbarker would enable a free alternative solution to use follow-me on
> mobile devices

Sorry, I wasn't clear.

If Tower already supports these messages then yes, supporting those 
messages in AP would allow a follow-me.

However, if Tower doesn't already support those then it could just as 
easily support follow-mode by broadcasting its position and setting the 
parameters appropriately.

lvale, 2018-08-21:
Sorry I wasn't clear also.
QGC supports this now, and it could be an alternative on mobile devices.

auturgy, 2019-01-27:
No reason to close. It’s a valid enhancement request.

rmackay9, 2019-09-30:
I think this is a duplicate of this other issue so let's close it: https://github.com/ArduPilot/ardupilot/issues/8517

closed 2019-09-30 by rmackay9