ArduPilot/ardupilot #3934: Using Dual GPS on pixhawk as compass

geofrancis, 2016-04-19:
i have been combing through the new dji a3 controller specs and seen that for the rtk gps they are using dual antennas as a redundant compass by using the different positions of each gps to calculate what direction the vehicle is pointing.

now that got me thinking about my pixhawk, it already has dual m8n gps units that are pretty accurate Its no rtk but i would think the accuracy of your heading would only limited by the separation of the gps units and accuracy of the gps.

jpkh, 2016-05-12:
To have gps based accurate heading measurement we really need sub-centimeter accuracy. Only rtk can provide close to accuracy what this type measurement needs. Having thick cloud at sky will lower down accuracy. So no, normal neo8 wont work. In generally gos accuracy is 1.5-5meters depensing on area.
Between highrise buildings even rtk wont work due multipathing and other interference. Compass is still our best way to detect headings.

geofrancis, 2016-05-25:
Now that cheap rtk gps units are becoming avalible it could be done for not a lot http://navspark.mybigcommerce.com/ns-hp-rtk-capable-gnss-receiver/

geofrancis, 2017-04-12:
Is there any progress on this? I know it was talked about and I had some questions.

The main reason for wanting this is for resistance to magnetic anomalies so I was wondering how the ekf3 would react if all the compasses went out but it still had a valid heading using the rtk Would it trigger an ekf failure or would it be able to continue a mission using the valid heading information from the rtk?

rmackay9, 2017-04-13:
No concrete progress on this.
Paul talked about it in his presentation at the un-conference last month (see blog posts on ardupilot.org).
I think realistically, it's going to require a company to sponsor this effort or another developer (i.e. not Paul) with strong EKF skills is going to need to step forward and take it on as a volunteer.

suraqis, 2017-11-17:
Why not to implement HDT NMEA message and leave the math to GNSS receiver with dual antenna heading? https://www.trimble.com/OEM_ReceiverHelp/V4.44/en/NMEA-0183messages_HDT.html

geofrancis, 2017-11-17:
that requires a very expensive gps receiver, the idea was to use 2 low cost rtk receivers to do the same thing.

piningforfjords, 2018-07-15:
I would be curious whether RTK is required.  I'm not sure the accuracy between two GPS receiving in the same vicinity would be as bad as 1.5 m.  (non expert opinion warning) I would expect them to be quite close as their signals would be passing through the same/similar distortions in the atmosphere.  A test would be easy with 2 vehicles and what their relative positions changes would be as you simultaneously  move them around on a plank.  My in interest in this is for search and rescue operation in the very high and very low altitudes where compass signals are weak in the horizontal axis.  IE near the poles.   If I was to get some preliminary data would Paul be able to give an estimate for the cost for the EKF implementation?

geofrancis, 2018-07-16:
that was my thoughts, your heading accuracy would be determined by the accuracy of the gps and the distance between the receivers. 

from my testing a m8n will usually be accurate to around 30cm so as long as they were over 1m apart  it should give reasonably accurate heading information.

proficnc, 2018-07-16:
From what I understand, Paul has a branch where he is working on this, I would love to see this in, even if it needs to be dual RTK.

Shrediquette, 2018-09-03:
I also have a lot of interest in this subject. So I am hoping to read some news here once available.

vincemic, 2018-10-11:
Current compass reliability is the achilles heel  of Ardupilot imo. There are three in my current setup and none of them work reliably for whatever reason. For example, yesterday I did a mission for over an hour  with no warnings or issues, then today with the same mission, equipment, and settings I can't get the compasses to work right at all. The only known difference is the temperature, 80 yesterday and 60 today.  GPS has been rock solid and my rover is a meter long so I think this might be a better solution.

reubenfinch, 2019-01-28:
What exactly is the problem with implementing this. Assuming someone is already running two RTK modules with the positional offsets correctly entered, then the code to determine heading from position is easy.

rmackay9, 2019-01-28:
@reubenfinch,

The core dev team always has a huge number of items on their to-do list (there are 1000+ enhancement requests on this list) so it can take time for a particular item to be addressed.  For individuals the best way to speed up resolution is to get involved with the development.  For companies they may choose to sponsor a developer to get a particular feature completed.

Part of the issue with this particular enhancement request is there is some doubt that it will work reliably with the L1 RTK GPSs that most users want to use.  If you're interested in contributing to getting this resolved, a good test would be to place a vehicle with RTK GPS outdoors, set LOG_DISARMED = 1 and after a minute or so rotate it 90 degrees, then download the dataflash log and manually calculate what the heading would be from the two GPS positions recorded.  I suspect the calculated heading won't be that great but if it is good then that might entice a developer into taking on this issue.

reubenfinch, 2019-01-28:
Sorry @rmackay9, my tone wasn't intended to be accusatory. I understand every user has their own wishlist.  :)  I have a couple of F9P's on preorder so I'll try that out when they arrive.

mwrightE38, 2019-02-05:
Just as an FYI thing this can be done outside of ardupilot with two piksi's   https://support.swiftnav.com/customer/portal/articles/2805901-piksi-multi---heading.

yota-code, 2019-03-14:
> To have gps based accurate heading measurement we really need sub-centimeter accuracy. Only rtk can provide close to accuracy what this type measurement needs. Having thick cloud at sky will lower [...]

You don't need RTK devices to assess the heading, but you may need the technologies RTK uses : ie. some way to analyze in real time the GPS carrier. And indeed, two GPS receiver won't do the job, but a single receiver with to antennas and the capability to process the raw GPS signal can. As this device, which is not RTK-enabled but which uses two antennas: https://www.navtechgps.com/vn_200_oem_gps_aided_inertial_navigation_system_gps/ins_/

auturgy, 2019-11-18:
Can this be closed? https://github.com/ArduPilot/ardupilot/pull/7215

kopterkraft, 2019-11-22:
As this seems to be the most active issue on that topic and other discussion link to this, I would ask to keep it open please.

I see two approaches:
1. Use two standardd GPS (or better: GNSS like M8P or ZED-F9P) and do the heading / yaw calculations from their raw data inside the flight controller (using the configuration data how the antennas are positioned.
2. Use a "ready to use, black box" solution, that is already built with two high accuracy antennas (that also may be RTK supported) and which is doing all the necessary calculations by itself. These units usually deliver the calculated heading as part of the NMEA sentences.

Examples for these kind of units:
https://support.swiftnav.com/customer/portal/articles/2805901-piksi-multi---heading
https://www.advancednavigation.com/product/gnss-compass

Assuming, you already get an accurate heading through the NMEA sentences (second approach), is there a configuration recommendation for ArduCopter 3.6.11 (which I am currently using) and later, which take into account these values with high priority (higher priority compared to the values calculated from the magnetometer raw data)?

piningforfjords, 2019-11-22:
Another option is by NovAtel.  They say they can spit out regular NMEA
strings and heading from their system.
I am also interested in this functionality.

proficnc, 2019-11-22:
This feature already exists using the F9P or HerePro

Any less accurate GPS would not be accurate enough for this

proficnc, 2019-11-22:
Sorry, wrong button

auturgy, 2020-01-22:
@tridge @rmackay9 can this be closed?
Yaw from rtk gps is now supported

AndKe, 2020-01-22:
please point me to some information about yaw from RTK. 
Is that like two M8P modules on the vehicle plus RTK from ground, or two M8P / M8T will work?
any configuration examples/documentation about the distance between antennas etc?

auturgy, 2020-01-22:
https://github.com/ArduPilot/ardupilot/pull/12844

More separation is better, minimum is ~20cm.

Regards,

James

rmackay9, 2020-02-06:
We support this feature now in master (this will go out with the 4.1 releases) so I will close this now.  We still need to document the setup: https://github.com/ArduPilot/ardupilot_wiki/issues/2509

closed 2020-02-06 by rmackay9