ArduPilot/ardupilot #3946: No --home flag for arducopter-quad, impossible to start

rpadovani, 2016-04-21:
#### Issue details
I build the arducopter with `waf`, but now I'm not able to start it.

Each time I try to start it I have this error:

`sudo ./build/linux/bin/arducopter-quad 

Init APM:Copter V3.4-dev

Free RAM: 262144
FW Ver: 120
----------------------------------------

load_all took 3us
ToneAlarm: Error!! please check if PWM overlays are loaded correctlyAP_Notify: Failed to initialise ToneAlarm0 0 0 DataFlash_File: buffer size=16384
�	Q}�COMPASS INIT ERROR
PANIC: AP_Baro::read unsuccessful for more than 500ms in AP_Baro::calibrate [2]
`

I had the same error when I build with make, but I used to fix it passing the --home flag to `./ArduCopter.elf`.

Unfortunately there is no such flag in the new bin, so I have no idea how to start ArduCopter. What am I missing?

Thanks in advance for your help!

#### Platform
- [ ] All
- [ ] AntennaTracker
- [x] Copter
- [ ] Plane
- [ ] Rover

guludo, 2016-04-21:
By your description, you want to use sitl. In that case you should
configure for sitl unread of linux:

```
waf configure --board sitl
```

Btw, this matter should be discussed on http://discuss.ardupilot.org :)

guludo, 2016-04-21:
s/unread/instead/

I'm using my phone hehe

rpadovani, 2016-04-23:
Right, thanks, I feel a bit stupid right now!

closed 2016-04-23 by rpadovani