ArduPilot/ardupilot #3950: RTL on failsafe untill critical event then land

WickedShell, 2016-04-22:
#### Issue details
On a long failsafe event (say GCS failsafe) the aircraft should RTL, and loiter at the RTL point until a critical event (such as low battery failsafe) at which point it should transition to a planned landing if one exists in the mission.

The use case is that with current master immediately transitioning to land logic on RTL is poor as the landing area may not be cleared, or you are trying to reacquire comms with the aircraft. In these cases its desirable to continue flying until continuing to do so would risk a crash.

#### Platform
- [ ] All
- [ ] AntennaTracker
- [x] Copter
- [x] Plane
- [ ] Rover

rmackay9, 2016-04-23:
There's a similar issue somewhere related to a two-stage failsafe.  I generally agree there's no need to immediately RTL in some cases although for Copter, currently the GCS failsafe is really a duplicate of the RC failsafe.

WickedShell, 2016-04-23:
Ah I had missed that one for copter regarding batteries. What I'm specifically looking for here is the RTL behavior (after the current set of events trigger it) that has a landing option at any point due to battery or time elapsed at the RTL point, but that doesn't immediately phase on to the landing until one of those conditions are met.

I guess this is actually very similar issues, I just hadn't found the other one before. I guess maybe I should close this in favor of #1759

R-Lefebvre, 2016-04-25:
This is a great idea for cases of RTL due to lost-comms.  It makes sense to loiter as long as possible before landing.

However, wouldn't this sort of accomplished with the RTL_LOIT_TIME paramter?  If it was set really long, would it hover until the actual battery FS went off?  Or would the battery failsafe also get stuck in Loiter and crash?

WickedShell, 2016-04-25:
I can't speak to copter but not on plane. Plane doesn't have a loiter timer, and the battery failsafe will only trigger a long failsafe, not any other special action.

lvale, 2016-04-25:
linked to this https://github.com/ArduPilot/ardupilot/issues/3772

rmackay9, 2016-04-26:
RTL_LOIT_TIME controls the amount of time that the vehicle hovers over home before descending.  It's a little used parameter that we could consider removing in place of a hard-coded time (like 3 seconds).

R-Lefebvre, 2016-04-26:
Why remove it?  Seems fine to just leave it in there?

lvale, 2016-04-26:
+1 with @R-Lefebvre 
When a drone comes to land, it's useful to have it loitering for some time to check if the landing area is clear of potential issues

ErikGroeneveld, 2016-04-26:
+1, on RTL due to power you would like to bring it down as fast as
possible, and hovering for 3 seconds may make it fall out of the sky..

Best,

Erik

het volgende geschreven:

lvale, 2016-04-26:
@ErikGroeneveld If the copter waits for 3s, at the default LAND_SPEED is a difference of 1.5m in height....If the drone has to fall from 15m (Default RTL Height) the difference is irrelevant IMHO.

ErikGroeneveld, 2016-04-26:
Hi Luis,

LAND_SPEED is only becoming active below 10m,

So if my RTL night is set at 20m, then for the first 10m wpnav_speed_dn is
used (by my understanding) and this is set at 300cm/s.

So without the delay at the RTL hight I am 9 meters lower in 3 seconds...

And like I said when you are in a failsave situation due to power, flying
back to your RTL location and bringing it down any second may make the
difference, between a crash or a save landing, so please don't hardcode
this useful setting...

I reposition in case needed on the way down...

Erik

lvale, 2016-04-26:
@ErikGroeneveld  If someone is flying where a 3 seconds delay (or other delay configurable like it is) would cause a brownout then perhaps using RTL might not be a good choice to save the drone. In the end we both have the same opinion of leaving it configurable:)

kudykam, 2016-06-29:
Two level battery failsafe would be great: for example: on 11V RTL then on 10.3 LAND.
Or LAND depends on quad height. If height <20m LAND at 10.2V, 20-50m land at 10.4V, 50-100m at 10.5V etc...

WickedShell, 2018-03-10:
Presented in #7213, tagging so I remember to close this.