ArduPilot/ardupilot #3951: make should throw an error if the binary size is to large

WickedShell, 2016-04-23:
#### Issue details
It's an edge case, but if you build a firmware that is to large to be uploaded to the board then make should throw an error and refuse to flash the binary (unless overridden with a flag). At the moment if you just do make px4-v2-upload it never prints the binary sizes at the end, making it easy to upload to large a binary.

#### Platform
- [X] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover

kwikius, 2016-05-05:
This should be caught by the linker. It will complain that the image wont fit I think and wont produce a binary

magicrub, 2016-05-27:
So this is specific to "px4-v2-upload" still attempting to upload?

WickedShell, 2016-05-27:
Yeah, or in general if the binary is to big we shouldn't end make/waf with
a success state