ArduPilot/ardupilot #3999: Building ArduPilot for Pixhawk/PX4 on Linux with Make

sgh055, 2016-04-28:
Hi, I have some problem.

I followed Building ArduPilot for Pixhawk/PX4 on Linux with Make(http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html). but When I make px4-v2, I got a error message.
How can i fix that? I installed all of them

This message.
[image]

kwikius, 2016-04-28:
It looks like arm-none-eabi-gcc which is the compiler for this board is not installed correctly. Check the bin directory the messages are telling you and see if there is a arm-none-eabi-gcc ( and other similar tools) in there

sgh055, 2016-04-28:
@kwikius 
[image]
I try to this command 'arm-none-eabi-gcc --version' but they say "no such file or directory"

however, there exist file.
[image]
what is mean??

WickedShell, 2016-04-28:
It's not being found in the search path. Check where you're $PATH points,
it needs to point into that directory for the tools to be found.

sgh055, 2016-04-28:
[image]
@WickedShell  you say this? ~/.profile

OXINARF, 2016-04-28:
I think you need to install libc6-i386

sgh055, 2016-05-01:
@OXINARF Thank you! OXINARF
It is working!

closed 2016-05-01 by sgh055