ArduPilot/ardupilot #4000: Switching to Guided Mode when hovering right above ground in loiter mode reduces motor speed and doesn't respond to waypoint

rrr6399, 2016-04-28:
I attempted the following steps:

1.) Arm
2..) Switch from stabilize to loiter
3.) Increase throttle to hover just above ground in loiter
4.) Switch to guided mode and send a msg_set_position_target_global_int with position and speed specified
Instead of moving the correct altitude specified by the msg_set_position_target_global_int, the power is reduced to idle and the copter drops to the ground.

#### Version
3.4 Arducopter firmware from April 4th.

#### Platform
- [ ] All
- [ ] AntennaTracker
- [x] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)
Hex

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk

rmackay9, 2016-04-28:
thanks for the report.   if you have a dataflash log that would help speed the analysis.

rrr6399, 2016-04-28:
@rmackay9 , I'll upload one soon. Did you get my logs for this issue:
https://github.com/ArduPilot/ardupilot/issues/3881

This bug was the result of trying to come up with a process to work around the 3881 issue. I'd rather use the takeoff command in guided mode, but its too risky until we know the underlying problem with it on the S900.

rrr6399, 2016-04-28:
After reviewing the logs, it looks like what may be going on is that the guided target is relative to the current loiter altitude which goes negative when the thrust is increased probably to prop wash near the ground. When guided reduces the throttle, the altitude paradoxically increases by 3 meters and probably fools the system into thinking it was actually at 3 meters above ground.

[image]

[image]

rrr6399, 2016-04-28:
Here's one of the log files.
[48.zip](https://github.com/ArduPilot/ardupilot/files/239756/48.zip)

rrr6399, 2016-04-28:
I'll run a test tomorrow where I change the target altitude to 6m.

rrr6399, 2016-04-28:
It appears that if the copter is not hovering stably in position using Loiter, that switching to guided mode kills the throttle.
[image]

[48.zip](https://github.com/ArduPilot/ardupilot/files/240959/48.zip)

rrr6399, 2016-04-28:
If I am hovering in stabilize mode and switch to guided, the copter falls out of the sky. That may be expected behavior though.

rrr6399, 2016-04-28:
If am Landing and switch to guided mode, it switches without incident and flys to the correct waypoint.

rrr6399, 2016-04-28:
Another odd behavior that I see is that when I switch from stabilize to loiter and do not change the throttle, the thrust drops, which requires me to increase the throttle to bring the copter back to altitude. Once the copter is stabilized in loiter and the indicated altitude is above 0, I can switch to guided successfully. See the end of the flight from this log:
[54.zip](https://github.com/ArduPilot/ardupilot/files/241041/54.zip)

rrr6399, 2016-04-28:
Here is the variation of alt and throttle while switching modes:
[image]

rrr6399, 2016-06-30:
This may be  related to https://github.com/ArduPilot/ardupilot/issues/4387

I'll verify soon.

rmackay9, 2016-07-01:
Ok, I think the drop in throttle is related to [this bug](https://github.com/ArduPilot/ardupilot/commit/d1b28aaed9f79ce5ce87a46eeed195e739b9e355) that's been fixed in master and will go out with -rc2.

OXINARF, 2016-11-02:
@rrr6399 This problem was found to be related to you using RC override only for some channels without using a RC receiver, right?

OXINARF, 2016-12-17:
No answer for one month and a half, closing.

closed 2016-12-17 by OXINARF