ArduPilot/ardupilot #4009: ArduPlane BIN logs are empty in V3.5.1+

t413, 2016-04-28:
Since V3.5.1 (working in V3.5.0) BIN log files stored on Pixhawks' SD cards have been zero-extended empty files.

Example of V3.5.0 log:
[6.BIN.zip](https://github.com/ArduPilot/ardupilot/files/241103/6.BIN.zip)

Example of V3.5.1 log:
[5.BIN.zip](https://github.com/ArduPilot/ardupilot/files/241110/5.BIN.zip)

Example of V3.5.2+ (latest release build):
[apm-3.5.3px4log-21.BIN.zip](https://github.com/ArduPilot/ardupilot/files/241111/apm-3.5.3px4log-21.BIN.zip)

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [x] Plane
- [ ] Rover

#### Airframe type
Multiple airframes. Flying wing and traditional.

#### Hardware type
This is the [PX4LITE-Pixhawk-Lite from banggood](http://www.banggood.com/PX4LITE-Pixhawk-Lite-32-Bit-Open-Source-Flight-Controller-for-RC-Multicopter-p-1009876.html).

ctonyperry, 2016-04-28:
Same issue, using a Pixhawk clone from Ebay:

http://www.ebay.com/itm/Pixhawk-PX4-2-4-6-32bit-Flight-Controller-Led-NEO-M8N-GPS-PPM-OSD-3DR-915Mhz-US-/251858758736?hash=item3aa3f3ac50:g:py4AAOSw-vlVpNjx

ctonyperry, 2016-04-28:
I am seeing the same issue on a pixhawk that I purchased from eBay. http://www.ebay.com/itm/Pixhawk-PX4-2-4-6-32bit-Flight-Controller-Led-NEO-M8N-GPS-PPM-OSD-3DR-915Mhz-US-/251858758736?hash=item3aa3f3ac50:g:py4AAOSw-vlVpNjx

peterbarker, 2016-04-28:
On Thu, 28 Apr 2016, ctonyperry wrote:

> I am seeing the same issue on a pixhawk that I purchased from eBay.http://www.ebay.com/itm/Pixhawk-PX4-2-4-6-32bit-Flight-Controller-Led-NEO-M
> 8N-GPS-PPM-OSD-3DR-915Mhz-US-/251858758736?hash=item3aa3f3ac50:g:py4AAOSw-v
> lVpNjx

Unable to reproduce on an old PX4 here and master.  Could you supply the 
git sha for the release you are using, please?

What was your log_bitmask parameter set to?

Does the file grow normally over time, and at the sort of rate you would 
expect given your log bitmask?

Have you tried a different SD card (two users having this problem, so it 
is quite doubtful the SD card is at fault).

What size SD card are you using?

(incidentally, I'm seeing those files as 0xFF filled, not 0-byte filled)

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

t413, 2016-04-30:
Using a different SD card I've confirmed it's a scam SD card, something I've actually experienced in cameras before but wasn't expecting here. Sorry for the erroneous bug!

Useful article for others who might have a similar problem: [How to check and spot fake Micro SD card ](http://www.happybison.com/reviews/how-to-check-and-spot-fake-micro-sd-card-8/)

closed 2016-04-30 by t413