ArduPilot/ardupilot #4016: Plane:elevon channel mixer's code seems not right

nocternal, 2016-04-30:
I fly skywaker X8 which needs elevon mixer,in [attitude.cpp](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/Attitude.cpp)
`void Plane::channel_output_mixer(uint8_t mixing_type, int16_t &chan1_out, int16_t &chan2_out) 
666 { 
667     int16_t c1, c2; 
668     int16_t v1, v2; 
669 
 
670     // first get desired elevator and rudder as -500..500 values 
671     c1 = chan1_out - 1500; 
672     c2 = chan2_out - 1500; 
......
`
c1 represent channel pitch radio out,for my rc,it's trim value is 1517.
c2 represent channel roll radio out,it's trim value is 1534.
if let  nav_pitch=0 and nav_roll=0 for example,ignore other compensation,finally my pitch channel radio out should be 1517 and channel roll radio out should be 1534  ,but use this mixer logic,my mixing gain is 1, and is in MIXING_UPDN ,then finally my left servo1(chan1) will get 1483 ,and servo2(chan2) will get 1449,that means my plane will rotate a little .
although pid controller can get feed back and make it fly right,but I think is it better to modify this mixer also consider input channel's value? not just let both of them minus 1500 to get new c1 and c2

WickedShell, 2016-04-30:
@magicrub If you want to take a look at this, it seems like a correct point, but I'm not familiar enough with the mixer to be confident on that. A quick glance and it looks like that is operating on the raw outputs, rather then a normalized range.

magicrub, 2016-04-30:
I'll look into this on Monday.

magicrub, 2016-05-01:
My initial thought is that we should never use a hard coded value "1500" in
the code and instead always use the trim value. I'll investigate. This is
probably a problem with all the mixers

Thanks @nocternal!

I'll look into this on Monday.

nocternal, 2016-05-01:
you are welcome~ expect new version mixer(^V^)

magicrub, 2016-05-01:
Expect? Does that mean you are working on a pull request?

nocternal, 2016-05-01:
nope,sorry,I'm not working on it...because I think you're right,maybe other mixers have same problem.There are lots of mixer ,,but I don't understand other mixers well,it's better for a more experienced man fix them I think(^_^)

nocternal, 2016-05-03:
How about standardize radio outs?Let calc pwm() sends out pwm radio out from 900 to 2100. In my opinion, it's a little weird that radio out and radio in use same trim and range ,is it better let radio outs trim standardized ,all at 1500? @magicrub ,if so, we needn't change mixers

kwikius, 2016-05-04:
FYI I did some work on a generic ArduPilot mixer, which creates a mixer form a script. It is designed to understand Ardupilot parameters and can use them as part of the mixer
https://github.com/kwikius/mixer_lang
An example script
https://github.com/kwikius/mixer_lang/blob/master/basic.mix

With that fleshed out all the current mixers would be redundant :)

magicrub, 2016-05-04:
fancy!

kwikius, 2016-05-05:
@magicrub 
Yes , though the mixer is not the trickiest part. That is to enter the labyrinth (of which the above issues are part) and slay the minotaur. :)

Naterater, 2019-01-04:
Although the logic in the first post may be sound, I think that this would have been an issue for more users by now. Also considering that this hasn't been revisited in almost 2.5 years, I think this should be closed.

closed 2019-01-04 by nocternal