ArduPilot/ardupilot #4018: waf: fails to compile on windows

WickedShell, 2016-05-01:
#### Issue details
waf crashes after compiling the mavlink bindings

#### Version
e7974702d4f75ca4c98058b104b492e778daff3c
Python 2.7.10
CYGWIN_NT-6.1-WOW milton 2.4.1(0.293/5/3) 2016-01-24 11:24 i686 Cygwin
Windows 7 Pro x64

#### Platform
- [x] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover
#### Logs
```
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light distclean
'distclean' finished successfully (0.052s)
 
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light copter
The project was not configured: run "waf configure" first!
 
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light configure
Setting top to                           : /home/WickedShell/ardupilot
Setting out to                           : /home/WickedShell/ardupilot/build
Setting board to                         : sitl
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for program 'python'                  : /usr/bin/python
Checking for python version                    : (2, 7, 10, 'final', 0)
Checking for program 'git'                     : /usr/bin/git
Benchmarks                                     : disabled
Unit tests                                     : enabled
'configure' finished successfully (2.219s)
 
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light copter
Waf: Entering directory `/home/WickedShell/ardupilot/build/sitl'
[3/3] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Validating ../../modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing ../../modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating ../../modules/mavlink/message_definitions/v1.0/common.xml
Parsing ../../modules/mavlink/message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FOLLOW_TARGET is longer than 64 bytes long (101 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message AUTOPILOT_VERSION is longer than 64 bytes long (68 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message V2_EXTENSION is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Merged enum MAV_CMD
Found 180 MAVLink message types in 2 XML files
Generating C implementation in directory /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega
Generating C implementation in directory /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
 
Waf: Leaving directory `/home/WickedShell/ardupilot/build/sitl'
Build failed
 -> task in 'mavlink' failed (exit status -6):
        {task 4288593068L: mavgen ardupilotmega.xml -> }
['/usr/bin/python', '/home/WickedShell/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py', '--lang=C', '--wire-protocol=1.0', '--output', '/home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0', '../../modules/mavlink/message_definitions/v1.0/ardupilotmega.xml']
 
WickedShell@milton ~/ardupilot
$

```

lucasdemarchi, 2016-05-06:
could you do the following?

1) run waf with `-v` option
2) run the following command in `/home/WickedShell/ardupilot/build/sitl`:

```
/usr/bin/python /home/WickedShell/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0 ../../modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
```

WickedShell, 2016-05-06:
Yeah, I'm in an airport at the moment, but I can easily run that this
evening.

WickedShell, 2016-05-07:
```
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light -v
Waf: Entering directory `/home/WickedShell/ardupilot/build/sitl'
00:55:41 runner ['/usr/bin/git', 'rev-parse', '--short=8', 'HEAD']
00:55:45 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/gtest')
00:55:47 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/mavlink')
[3/4] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
[4/4] Creating build/sitl/ap_version.h
00:55:49 runner ['/usr/bin/python', '/home/WickedShell/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py', '--lang=C', '--wire-protocol=1.0', '--output', '/home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0', '../../modules/mavlink/message_definitions/v1.0/ardupilotmega.xml']
Validating ../../modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing ../../modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating ../../modules/mavlink/message_definitions/v1.0/common.xml
Parsing ../../modules/mavlink/message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FOLLOW_TARGET is longer than 64 bytes long (101 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message AUTOPILOT_VERSION is longer than 64 bytes long (68 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message V2_EXTENSION is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Merged enum MAV_CMD
Found 180 MAVLink message types in 2 XML files
Generating C implementation in directory /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega
Generating C implementation in directory /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers

Waf: Leaving directory `/home/WickedShell/ardupilot/build/sitl'
Build failed
 -> task in 'mavlink' failed (exit status -6):
        {task 4288105964L: mavgen ardupilotmega.xml -> }
['/usr/bin/python', '/home/WickedShell/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py', '--lang=C', '--wire-protocol=1.0', '--output', '/home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0', '../../modules/mavlink/message_definitions/v1. [... cut ...]
Note: message V2_EXTENSION is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Merged enum MAV_CMD
Found 180 MAVLink message types in 2 XML files
Generating C implementation in directory /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega
Generating C implementation in directory /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
Aborted (core dumped)

WickedShell@milton ~/ardupilot/build/sitl
$

```

meee1, 2016-05-09:
return code on error is
Note: message AUTOPILOT_VERSION is longer than 64 bytes long (68 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message V2_EXTENSION is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Merged enum MAV_CMD
Found 180 MAVLink message types in 2 XML files
Generating C implementation in directory /tmp/ArduPlane.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega
Generating C implementation in directory /tmp/ArduPlane.build/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
Aborted (core dumped)

michael@WIN-3NBOUTHN4TN /cygdrive/c/Users/michael/Desktop/DIYDrones/ardupilot/ArduPlane
$ echo $?
134

meee1, 2016-05-09:
additional
Found 180 MAVLink message types in 2 XML files
Generating C implementation in directory /tmp/ArduPlane.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega
Generating C implementation in directory /tmp/ArduPlane.build/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
last line in mavgen.py
Aborted (core dumped)

michael@WIN-3NBOUTHN4TN /cygdrive/c/Users/michael/Desktop/DIYDrones/ardupilot/ArduPlane
$ echo $?
134

the crash is in python, not from the script itself. I added a print to the end of mavgen.py.

$ uname -a
CYGWIN_NT-10.0-WOW WIN-3NBOUTHN4TN 2.0.4(0.287/5/3) 2015-06-09 12:20 i686 Cygwin

lucasdemarchi, 2016-05-10:
@meee1 @WickedShell completely untested in a machine that crashes, but could you give it a try? https://github.com/lucasdemarchi/ardupilot/tree/pr/waf-windows-workaround

WickedShell, 2016-05-10:
Still no joy
```
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light configure
Setting top to                           : /home/WickedShell/ardupilot
Setting out to                           : /home/WickedShell/ardupilot/build
Setting board to                         : sitl
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for program 'python'                  : /usr/bin/python
Checking for python version                    : (2, 7, 10, 'final', 0)
Checking for program 'git'                     : /usr/bin/git
Benchmarks                                     : disabled
Unit tests                                     : enabled
'configure' finished successfully (8.723s)

WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light -v
Waf: Entering directory `/home/WickedShell/ardupilot/build/sitl'
19:47:42 runner ['/usr/bin/git', 'rev-parse', '--short=8', 'HEAD']
19:47:45 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/gtest')
19:47:47 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/mavlink')
[4/4] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
[4/4] Creating build/sitl/ap_version.h
19:47:48 runner '/usr/bin/python /home/WickedShell/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0 /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml'
Validating /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Parsing /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FOLLOW_TARGET is longer than 64 bytes long (101 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
[... cut ...]
$ git status
HEAD detached at lucasdemarchi/pr/waf-windows-workaround
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   mk/board_native.mk

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        ArduPlane/python2.7.exe.stackdump

no changes added to commit (use "git add" and/or "git commit -a")

WickedShell@milton ~/ardupilot
$
```

lucasdemarchi, 2016-05-10:
@WickedShell ok... it seems python uses signed. Just pushed a version that may work. Could you try again?

WickedShell, 2016-05-10:
@lucasdemarchi I'm always happy to test, still seems to be failing

```
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light configure
Setting top to                           : /home/WickedShell/ardupilot
Setting out to                           : /home/WickedShell/ardupilot/build
Setting board to                         : sitl
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for program 'python'                  : /usr/bin/python
Checking for python version                    : (2, 7, 10, 'final', 0)
Checking for program 'git'                     : /usr/bin/git
Benchmarks                                     : disabled
Unit tests                                     : enabled
'configure' finished successfully (2.159s)

WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light -v
Waf: Entering directory `/home/WickedShell/ardupilot/build/sitl'
20:37:17 runner ['/usr/bin/git', 'rev-parse', '--short=8', 'HEAD']
20:37:18 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/gtest')
20:37:19 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/mavlink')
[4/4] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
[4/4] Creating build/sitl/ap_version.h
20:37:20 runner '/usr/bin/python /home/WickedShell/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0 /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml'
Validating /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Parsing /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FOLLOW_TARGET is longer than 64 bytes long (101 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
[... cut ...]
Generating C implementation in directory /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers

Waf: Leaving directory `/home/WickedShell/ardupilot/build/sitl'
Build failed
Traceback (most recent call last):
  File "/home/WickedShell/ardupilot/modules/waf/waflib/Task.py", line 217, in process
    ret = self.run()
  File "Tools/ardupilotwaf/mavgen.py", line 29, in run
    Logs.warning('mavgen crashed with code: {}'.format(ret))
AttributeError: 'module' object has no attribute 'warning'
```

OXINARF, 2016-05-10:
@WickedShell actually tridge merged this and fixed that error, can you test master?

WickedShell, 2016-05-10:
@OXINARF still fails

```
WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light configure
Setting top to                           : /home/WickedShell/ardupilot
Setting out to                           : /home/WickedShell/ardupilot/build
Setting board to                         : sitl
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for program 'python'                  : /usr/bin/python
Checking for python version                    : (2, 7, 10, 'final', 0)
Checking for program 'git'                     : /usr/bin/git
Benchmarks                                     : disabled
Unit tests                                     : enabled
'configure' finished successfully (2.255s)

WickedShell@milton ~/ardupilot
$ ./modules/waf/waf-light -v
Waf: Entering directory `/home/WickedShell/ardupilot/build/sitl'
20:47:53 runner ['/usr/bin/git', 'rev-parse', '--short=8', 'HEAD']
20:47:54 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/gtest')
20:47:55 runner ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/WickedShell/ardupilot/modules/mavlink')
[3/4] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
[4/4] Creating build/sitl/ap_version.h
20:47:57 runner '/usr/bin/python /home/WickedShell/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output /home/WickedShell/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v1.0 /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml'
Validating /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Parsing /home/WickedShell/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FOLLOW_TARGET is longer than 64 bytes long (101 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
[... cut ...]
  2. 'ArduPlane_libs' in /home/WickedShell/ardupilot/ArduPlane
* Node /home/WickedShell/ardupilot/build/sitl/libraries/AP_RPM/RPM_SITL.cpp.4.o is created more than once (full message on 'waf -v -v'). The task generators are:
  1. 'ArduCopter_libs' in /home/WickedShell/ardupilot/ArduCopter
  2. 'ArduCopter_libs' in /home/WickedShell/ardupilot/ArduCopter
Waf: Leaving directory `/home/WickedShell/ardupilot/build/sitl'
Build failed
 -> task in 'mavlink' failed (exit status -6):
        {task 4287689708L: mavgen ardupilotmega.xml -> }
''

```

OXINARF, 2016-05-10:
@WickedShell ah just noticed, tridge didn't merge the second version Lucas put, but the first.

WickedShell, 2016-05-10:
@OXINARF since all of Lucases branches fail for me I'm not to surprised master failed as well :)

tridge, 2016-05-10:
i've pushed some changes which make it pass for both my windows boxes now, at least for building SITL

OXINARF, 2016-05-10:
@tridge I made comments in the commits: https://github.com/ArduPilot/ardupilot/commit/5ef0c4d3fe083e67a084f166f6d1a9070c365c3b#commitcomment-17411303 and https://github.com/ArduPilot/ardupilot/commit/9653aae21f4047140026bca8b660f0edd028f059#commitcomment-17411315

WickedShell, 2016-05-10:
master as of ~2 minutes ago made it through mavlink, still compiling the rest now, (I did see an error code, but it continued past it)

lucasdemarchi, 2016-05-10:
ok... seems to be working now. Thanks.

closed 2016-05-10 by lucasdemarchi