ArduPilot/ardupilot #4019: error terrain alt variance

onurboy, 2016-05-01:
#### Issue details
_Sometimes, when I am flying with my pixhawk installed skywalker 1880 plane, I get error terrain alt variance message, EKF terrain bar fully red and it continues till the end of the flight especially in hard bank turns. Same day with several flights maybe only 2 or 3 of them or some times none of them give this error. Do you have an idea whats the root cause? 
Thank you,

Onur.

#### Version
_Plane 3.5.0

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [ x] Plane
- [ ] Rover

#### Airframe type
_Skywalker 1880 fixed wing_

#### Hardware type
_pixhawk (pixhawk, pixracer, PX4FMU etc)_

#### Logs
[https://www.dropbox.com/s/nsoarh7rdviyohr/2016-04-30%2011-26-59.bin?dl=0](url)

WickedShell, 2016-05-01:
This is an airspeed variance rather then terrain. It has to do with a
mismatch between the ekf and mission planner. It means there was a
difference between the estimated airspeed and the reported airspeed. There
is an issue tracking that already open

OXINARF, 2016-05-01:
@WickedShell are you talking about #3569?

WickedShell, 2016-05-01:
Yeah, I'm in a airport and hadn't searched the issues list.

OXINARF, 2016-05-01:
No problem. From that issue I gather that starting with Plane 3.5.1 it should be fixed right?

WickedShell, 2016-05-01:
Yeah, this can be closed as it has been addressed. I don't none of Paul
would want to look at the log to look at why the airspeed warnings
occurred. EKF2 tightened the tolerance levels for this, and I suspect that
is the root cause of this.

OXINARF, 2016-05-02:
@onurboy This has been fixed starting with Plane 3.5.1, so you should update. Also next time, please post first in the forums (http://discuss.ardupilot.org), this issues list should be only for confirmed bugs.

@priseborough Tagging you just in case you want to look at the log.

closed 2016-05-02 by OXINARF