ArduPilot/ardupilot #4023: waf: fail to update submodule

lucasdemarchi, 2016-05-02:
#### Issue details
It's not possible to update submodule

#### Platform
- [x] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover

lucasdemarchi, 2016-05-02:
Move issue out of #3932 to be tracked here. Relevant details below.

lucasdemarchi, 2016-05-02:
```
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$ git submodule update --init
Submodule 'modules/PX4Firmware' () registered for path 'modules/PX4Firmware'
Submodule 'modules/PX4NuttX' () registered for path 'modules/PX4NuttX'
Submodule 'modules/gbenchmark' () registered for path 'modules/gbenchmark'
Submodule 'gtest' () registered for path 'modules/gtest'
Submodule 'modules/mavlink' () registered for path 'modules/mavlink'
Submodule 'modules/uavcan' () registered for path 'modules/uavcan'
Submodule 'modules/waf' () registered for path 'modules/waf'
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$ git submodule status
 7fd7b2b1b0ecdad69d92e0bf0abcb7849d3826a1 modules/PX4Firmware (stable_manual_flight-15918-g7fd7b2b)
 579e82d4567e0884f113447d255d8e8e4335ac76 modules/PX4NuttX (ArduCopter-beta-19-g579e82d)
 006d23ccca1375a973b7fae0cc351cedb41b812a modules/gbenchmark (v0.1.0)
 c99458533a9b4c743ed51537e25989ea55944908 modules/gtest (release-1.7.0)
 ba2dd74d976f6c15291079608c5b00f39fc2463d modules/mavlink (1.0.9-1211-gba2dd74)
 7ce96d6c1e5aeb82fa90ade26a49552c122a984b modules/uavcan (APMrover2-2.49-819-g7ce96d6)
 7dee3bdb8f30398621dd3764f969903b1bc68850 modules/waf (waf-1.8.18-2-g7dee3bd)
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$ alias waf=$PWD/modules/waf/waf-light
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$ waf configure --board=linux
Setting top to                           : /home/andy/cpp/projects/apmwaf/ardupilot 
Setting out to                           : /home/andy/cpp/projects/apmwaf/ardupilot/build 
Setting board to                         : linux 
Checking for 'g++' (C++ compiler)        : /usr/bin/g++ 
Checking for 'gcc' (C compiler)          : /usr/bin/gcc 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version                    : (2, 7, 3, 'final', 0) 
Checking for program 'git'                     : /usr/bin/git 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
'configure' finished successfully (6.563s)
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$ waf check
Waf: Entering directory `/home/andy/cpp/projects/apmwaf/ardupilot/build/linux'
Waf: Leaving directory `/home/andy/cpp/projects/apmwaf/ardupilot/build/linux'
Build failed
Traceback (most recent call last):
  File "/home/andy/cpp/projects/apmwaf/ardupilot/modules/waf/waflib/Runner.py", line 294, in task_status
    return tsk.runnable_status()
  File "Tools/ardupilotwaf/git_submodule.py", line 48, in runnable_status
    out = self.generator.bld.cmd_and_log(cmd, quiet=Context.BOTH)
  File "/home/andy/cpp/projects/apmwaf/ardupilot/modules/waf/waflib/Context.py", line 459, in cmd_and_log
    raise e
WafError: Command ('/usr/bin/git', '-C', '/home/andy/cpp/projects/apmwaf/ardupilot', 'submodule', 'status', '--recursive', '--', '/home/andy/cpp/projects/apmwaf/ardupilot/modules/gtest') returned 129

andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$

```

lucasdemarchi, 2016-05-02:
@kwikius it seems that your git program doesn't recognize an option we are passing. I'd bet it doesn't have the -C option, in which case we should provide a fallback and just do the right thing. What's your git version?

And... could you run this manually?:
```
/usr/bin/git -C /home/andy/cpp/projects/apmwaf/ardupilot submodule status --recursive -- /home/andy/cpp/projects/apmwaf/ardupilot/modules/gtest
```

kwikius, 2016-05-03:
@lucasdemarchi 

Here is the output
```
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$ /usr/bin/git -C /home/andy/cpp/projects/apmwaf/ardupilot submodule status --recursive --home/andy/cpp/projects/apmwaf/ardupilot/modules/gtest
Unknown option: -C
usage: git [--version] [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
           [-p|--paginate|--no-pager] [--no-replace-objects] [--bare]
           [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
           [-c name=value] [--help]
           <command> [<args>]
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$ git --version
git version 1.7.9.5
andy@andy-netbook:~/cpp/projects/apmwaf/ardupilot$
```

OXINARF, 2016-05-03:
@kwikius That's quite an old git version, 4 years. The -C option was introduced in git 1.8.5, which was released 2.5 years ago.
@lucasdemarchi I think we should support different versions of tools but obviously we need limits, and, for me, this one passes the limit.

kwikius, 2016-05-03:
My version of Ubuntu is 12.04. This is the default version of git I get via the package manager. This version of Ubuntu is still current..
https://wiki.ubuntu.com/Releases
Sure I could install a newer version of git but now there is another dependency required by waf which wasnt needed with make. Whose life is waf simplifying then and what is the point of it?  It  seems it is raising the barrier to entry for others trying to build the Ardupilot code.

lucasdemarchi, 2016-05-03:
I think the change is simple enough to support it. If it gives more
problems we can revisit the requirements and check the minimum version in
the configure step.

Btw this is not a make vs waf difference. It's just regarding the way an
*external* tool is called.

lucasdemarchi, 2016-05-03:
Let me know if this is still an issue for you.

closed 2016-05-03 by lucasdemarchi

kwikius, 2016-05-04:
It is working for me now. Thanks

josheeg, 2018-11-27:
sudo apt install git-all

josheeg, 2018-11-27:
how do I update waf? How should I install it?

lucasdemarchi, 2018-11-27:
Please see build instructions. You should NOT install waf, it's a local command.

zhangyihangceshi1hao, 2024-08-18:
Build failed
Traceback (most recent call last):
  File "/cygdrive/e/UAV/ardupilot/modules/waf/waflib/Runner.py", line 395, in task_status
    return tsk.runnable_status()
  File "/cygdrive/e/UAV/ardupilot/Tools/ardupilotwaf/git_submodule.py", line 74, in runnable_status
    out = self.generator.bld.cmd_and_log(cmd, quiet=Context.BOTH, cwd=self.cwd)
  File "/cygdrive/e/UAV/ardupilot/modules/waf/waflib/Context.py", line 461, in cmd_and_log
    raise e
waflib.Errors.WafError: Command ('/cygdrive/c/Program Files/Git/cmd/git', 'submodule', 'status', '--recursive', '--', '/cygdrive/e/UAV/ardupilot/modules/ChibiOS') returned 128