ArduPilot/ardupilot #4026: Compilation failure with gcc 5.3

peterbarker, 2016-05-02:
#### Issue details

Compilation fails because isnan and isinf are not in the global namespace.  A simple "make -j8 linux VERBOSE=1" is sufficient to reproduce (so long as you're using gcc 5.3.1)
 
> In file included from /home/pbarker/rc/ardupilot/libraries/AP_Common/Location.h:11:0,
>                  from /home/pbarker/rc/ardupilot/libraries/AP_Common/Location.cpp:5:
> /home/pbarker/rc/ardupilot/libraries/AP_Math/AP_Math.h: In function 'float constrain_float(float, float, float)':
> /home/pbarker/rc/ardupilot/libraries/AP_Math/AP_Math.h:80:15: error: 'isnan' was not declared in this scope
>   if (isnan(amt)) {
>                ^
> compilation terminated due to -Wfatal-errors.

#### Version

github master (384ad94e73d7decd2958880f2f91ba3324c96ba6)

#### Platform
- [ x] SITL
- [ x] linux

#### Airframe type
n/a

#### Hardware type
n/a

#### Logs
n/a

The following patch is sufficient to fix the problem, but probably needs to wrap the using std:: lines in some sort of #ifdef for building under make

> diff --git a/libraries/AP_Common/missing/cmath b/libraries/AP_Common/missing/cmath
> index 2709fd5..f95ccc4 100644
> --- a/libraries/AP_Common/missing/cmath
> +++ b/libraries/AP_Common/missing/cmath
> @@ -13,3 +13,6 @@ using std::isinf;
>  #if defined(HAVE_CMATH_ISNAN) && defined(NEED_CMATH_ISNAN_STD_NAMESPACE)
>  using std::isnan;
>  #endif
> +
> +using std::isnan;
> +using std::isinf;
>

OXINARF, 2016-05-03:
As has been announced before we will be changing to the waf build system for Linux (and SITL).

The correct fix for the make build system is at https://github.com/ArduPilot/ardupilot/pull/3878#issuecomment-207663453
The reason it's not on master is because it would break older compilers.

I'll leave this open until we change to waf since we have received some issues about it and an open issue is easier for people to find.

peterbarker, 2016-05-03:
@OXINARF What are your thoughts on this? https://github.com/ArduPilot/ardupilot/compare/master...peterbarker:recent-gcc-and-isinf?expand=1

OXINARF, 2016-05-03:
@peterbarker This is actually an issue with glibc 2.23 but that's usually very tied with gcc 5.3 so I don't have an issue with that fix. Let me tag @guludo and @lucasdemarchi since they are the maintainers of the build system.

lucasdemarchi, 2016-05-14:
I don't object to it. But I must say this is a very fragile check likely to break soon.

magicrub, 2016-05-27:
is this ready to be closed?

lucasdemarchi, 2016-05-27:
yes

closed 2016-05-27 by lucasdemarchi